chundermike/rpi-fruitbox-v2 #1: Error during the installation

DPAZ84, 2022-06-01:
Hi Mike, first of all that is incredible the work that you have been doing for the community. I was trying to install the Fruitbx v2 and I came across with an error at the end related to I have a Raspberry Pie 3, instead of a Pie 4 or something like that I can't remember the entire sentence. As far as I know this should work in Pie3

DPAZ84, 2022-06-01:
[image]

DPAZ84, 2022-06-01:
I was pretty sure that's something wrong from my side specially when I quite new coding or in the world of raspberry pi. Might I ask you how I can update the OS? I have installed RetroPie and emulation station. I want to add this as another functionality. That being said if I update the OS Can I jeopardize what is already installed?(RetroPie)
Really appreciate you time 
And all your responses

chundermike, 2022-06-01:
Hi

I'd recommend that you use a different SD card with just fruitbox on it. If you change the OS with the one needed for fruitbox then you will destroy your RetroPie image.

So take a fresh blank SD card, download the OS zip file to a PC from the link I sent, then burn the SD card with it using the raspberry pi imaging utility (from the Raspberry Pi website).

Put the SD card into your Pi and power up. After logging in as user pi (password raspberry), type sudo raspi-config

From here set up your WiFi, turn on SSH and change the GPU memory to 256.  Now you should be able to install fruitbox as normal.  When you want to copy your mp3 files to the Pi from your PC, you can use a PC utility called WinSCP which will allow you to drag and drop files between your PC and Pi. (The Pi connection is called raspberrypi.local)

Cheers
Mike

DPAZ84, 2022-06-01:
Thanks to god I asked you first!! 
I will use a new SD card install fruitbox and then install again retropie with emulators and rooms. 
Last question I promise you haha
Once I installed Fruitbox and RetroPie, It's possible to select which software boot initially. For example root into RetroPie and then select run Fruitbox.

chundermike, 2022-06-02:
Hi,

Not 100$ sure about this (it's not fruitbox related). Probably best to Google it!

Cheers
Mike

closed 2022-06-02 by chundermike