chundermike/rpi-fruitbox-v2 #3: 2 zero delay usb encoder

Simon410, 2022-06-18:
Hello Mike,
thanks for your wonderfull work on the version V2.01.
I have a problem with my 2 encoders, when i program them from the menu the buttons are recognised and accepted.
When i restart, the buttons won't correspond anymore or wan't work.
P.S. changed file to txt to upload.
Greetings,
John
[fruitbox.ini.txt](https://github.com/chundermike/rpi-fruitbox-v2/files/8933096/fruitbox.ini.txt)

Simon410, 2022-06-18:
Here are some screenshots of my project for a starlet jukebox.

Greetings,
John
[image]
[image]
[image]

chundermike, 2022-06-18:
Hi John

Skin looks fab by the way.

Regarding the encoders, this is a known issue I was told about last week, and I'm working on a fix. The problem is when you have more than one identical input device, in which case fruitbox can't distinguish between them. Assuming this bug is causing your issues. The fix will be in the next release and hopefully it'll resolve your problem.

Cheers
Mike

Simon410, 2022-06-18:
Thanks Mike for the quick reply, it worked with the V1.6 .
When this is solved and all works perfect i will share my setup.

Greetings,
John

chundermike, 2022-06-18:
By the way, if you're still working on your skin then ignore the next comment, but I noticed your song text (artist name) isn't centred vertically on the title strips, and long names don't fit.  All these things can be corrected by tweaking the ArtistText (and SongText, etc.) parameters.  Check out the user guide section 12.6.

Simon410, 2022-06-18:
Ok Mike, work in progress, needed to resize the screen.

Greetings,
John

Simon410, 2022-06-18:
Hi Mike, i found the solution on Reddit: Add the vendor type at the end of the cmdline.txt file "usbhid.quirks=0x0079:0x0006:0x40" and a a space at the end of the text and save.

Reddit : https://www.reddit.com/r/RetroPie/comments/bf8wal/need_help_with_2_zero_delay_usb_encoders_working/

Greetings,
John

closed 2022-07-03 by chundermike

saarfranzose, 2022-07-03:
Hello, I also use the no-delay-encoder for song-selecting in combination with the SmallWall-Skin. It is a really great software! I found the BTN-names in der the txt-file here in post#1:

  Select1 = BTN_BASE
  Select2 = BTN_BASE2
  Select3 = BTN_BASE3
  Select4 = BTN_BASE4

but where can I found a complete list of the possible BTN-Names? Is it possible to assign the no-delay-joystick?

Greetings from Germany and thank you!

chundermike, 2022-07-03:
Hi

Just assign the keys in the on-screen menu (press TAB), then select Settings then Configure Buttons. Scroll down to the button you want to assign, press select (ENTER on the keyboard by default), then press the key you want on your zero delay encoder or joystick.

Cheers
Mike

saarfranzose, 2022-07-03:
thank you Mike! That works! Althought there is no chance zu scroll up and down with a keyboard (because ist is a german keyboard?) But I could use a touchscreen to move within the menu.
I also tried using the Hori Controller "Real Arcade Pro.3", and it also works! I'm excited!

saarfranzose, 2022-07-04:
Hello Mike,

 

want to show you my fruitbox-description and my projekts in a german radio-forum. Still working on it.

 

https://radio-bastler.de/forum/showthread.php?tid=20130

 

https://radio-bastler.de/forum/showthread.php?tid=20296

 

have a nice week,

Jupp

chundermike, 2022-07-05:
Hi,

Thank you, that is a great article. You have a very good knowledge now on fruitbox.

One quick thing...you mentioned that for radio skins you need the skip button to stop the current playing stream before starting a new one.  Actually fruitbox v2 has a new feature called instant song play mode which overcomes this limitation. Basically it will stop any currently playing stream or song immediately a new song is selected.  This can be enabled in the skin file by setting InstantPlay = Yes in the [general] section.

Cheers
Mike

https://radio-bastler.de/forum/showthread.php?tid=20130

https://radio-bastler.de/forum/showthread.php?tid=20296

saarfranzose, 2022-07-05:
Thank you for the hint, Mike!

 

Something’s going wrong while Installation V2.02. 

 

Don’t want to steal your time. Here is my report, if you want to see it.

 

Prepare for a clean Installation:

 

***@***.***:~ $ sudo rm fruitbox -r

***@***.***:~ $ ls

install_fruitbox  LCD-show  main.1  Music

***@***.***:~ $ sudo rm main.1

***@***.***:~ $ sudo rm install_fruitbox

***@***.***:~ $ sudo rm /opt/vc/lib/libEGL.so

***@***.***:~ $ sudo rm /opt/vc/lib/libGLESv2.so

***@***.***:~ $ sudo rm /opt/vc/lib/libOpenVG.so

***@***.***:~ $ sudo rm /opt/vc/lib/libWFC.so

***@***.***:~ $ ls

LCD-show  Music

***@***.***:~ $

 

wget https://github.com/chundermike/rpi-fruitbox-v2/raw/main/install_fruitbox

…

2022-07-05 12:27:42 (3.58 MB/s) - ‘install_fruitbox’ saved [30052/30052]

chmod +x ./install_fruitbox

./install_fruitbox

 

…….

mv: cannot stat '/home/pi/fruitbox': No such file or directory

**********************************************************************

WARNING : Couldn't backup ~/fruitbox to ~/fruitbox_backup

ln: failed to create symbolic link '/home/pi/fruitbox/rpi/fruitbox': File exists

**********************************************************************

ERROR : Couldn't create softlink to ~/fruitbox/rpi/fruitbox

**********************************************************************

 

***@***.***:~ $ cd fruitbox

***@***.***:~/fruitbox $ ls

install_fruitbox  README.md  release_notes.txt  rpi  rpi4  skins  userguide.txt

***@***.***:~/fruitbox $ ls rpi

fruitbox

***@***.***:~/fruitbox $

 

***@***.***:~/fruitbox $ sudo cp rpi/fruitbox ./

 

sudo chmod +x ./fruitbox

 

the box runs, Mouse runs, but keyboard doesn’t. Fruitbox.ini seems okay, attract mode plays music.

 

Raspberry Pi 3 Modell B 1.2, Sound and Video over HDMI.

 

Have a good day,

Jupp

 

https://radio-bastler.de/forum/showthread.php?tid=20130

https://radio-bastler.de/forum/showthread.php?tid=20296

saarfranzose, 2022-07-05:
Hi Mike,

 

something strange .. fruitbox don’t accept official Raspberry keyboard , but a small Bluetooth Keyboard works!

 

greetings

Jupp

 

https://radio-bastler.de/forum/showthread.php?tid=20130

https://radio-bastler.de/forum/showthread.php?tid=20296

chundermike, 2022-07-05:
Hi Jupp

Thanks for spotting that – it’s a bug.  Now fixed in v2.03.

Cheers,
Mike

https://radio-bastler.de/forum/showthread.php?tid=20130

https://radio-bastler.de/forum/showthread.php?tid=20296

beit-no1, 2022-07-05:
Just tried updating my installation that was v2.01 by downloading and running the new install_fruitbox while ssh'd into the RaspBerry pi 3b.  Why the following error... Fruitbox is not currently running either

....
Saving to: ‘main.1’
main.1                                  [                                               <=>                 ]  84.86M   592KB/s    in 2m 8s
2022-07-05 23:40:40 (682 KB/s) - ‘main.1’ saved [88986846]

ln: failed to create symbolic link '/opt/vc/lib/libEGL.so': File exists
**********************************************************************
ERROR : Couldn't create softlink
**********************************************************************

beit-no1, 2022-07-05:
Tried again and this time rm main.1 and rm main before starting as I was also rm install_fruitbox to assure it does not create the file.1 version.  Still similar error.  Not sure if related to the v2.03 as I had it last night with v2.02

Connecting to codeload.github.com (codeload.github.com)|140.82.113.10|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘main’

main                                    [                                            <=>                    ]  84.86M   448KB/s    in 1m 47s

2022-07-05 23:58:49 (814 KB/s) - ‘main’ saved [88986846]

ln: failed to create symbolic link '/opt/vc/lib/libEGL.so': File exists
**********************************************************************
ERROR : Couldn't create softlink
**********************************************************************

chundermike, 2022-07-05:
Hi

You are attempting a full install when a previous install has occurred, so the softlinks to libEGL etc. are still there. I will modify the install utility to cope with these errors and that should hopefully fix the issue.

Cheers
Mike

beit-no1, 2022-07-05:
I thought I was following the upgrade instructions from v2.01 where the update in the settings is not working

cd /home/pi/fruitbox
wget https://github.com/chundermike/rpi-fruitbox-v2/raw/main/install_fruitbox
chmod +x ./install_fruitbox
rm install_fruitbox (added this to clear and avoid install_fruitbox.1)
rm main (added this to clear and avoid main.1)
./install_fruitbox

Will watch for new version

chundermike, 2022-07-05:
Hi,

That’s right, but to just update and not a full install you need...

./install_fruitbox --update

The new version (v1.3) of install_fruitbox is now on GitHub.

Cheers,
Mike

beit-no1, 2022-07-05:
you of course are right

now getting this... after the new wget
 chmod +x ./install_fruitbox
pi@raspberrypi:~/fruitbox $ ./install_fruitbox
-bash: ./install_fruitbox: cannot execute binary file: Exec format error

beit-no1, 2022-07-05:
dir -l

-rwxr-xr-x  1 root root 7778640 Jun 12 01:16 fruitbox
lrwxrwxrwx  1 pi   pi        30 May 31 04:29 fruitbox_2.00 -> /home/pi/fruitbox/rpi/fruitbox
-rw-r--r--  1 pi   pi      2547 Jul  5 02:58 fruitbox.ini
-rw-r--r--  1 pi   pi     10096 Jul  5 03:02 fruitbox.log
-rw-r--r--  1 pi   pi      2566 May 31 19:56 fruitbox.v2ini
-rwxr-xr-x  1 pi   pi     32312 Jul  6 00:46 install_fruitbox
-rw-r--r--  1 pi   pi      2036 May 30 22:59 README.md
-rw-r--r--  1 pi   pi     16803 Jul  5 03:01 release_notes.txt
drwxr-xr-x  2 pi   pi      4096 May 30 22:59 rpi
drwxr-xr-x 19 pi   pi      4096 May 30 22:59 skins
-rw-r--r--  1 pi   pi    121554 May 30 22:59 userguide.txt

chundermike, 2022-07-06:
Ah, I built install_fruitbox on a Pi4...seems like it produced a binary incompatible with the Pi3.  I’ve now created two different install_fruitbox executables, one for Pi3 (and below) and one for Pi4.

Mike

beit-no1, 2022-07-06:
./install_fruitbox --update
**********************************************************************
fruitbox installer v1.3 (update only)
**********************************************************************
Raspberry Pi Hardware version 3 detected
--2022-07-06 01:04:53--  https://raw.githubusercontent.com/chundermike/rpi-fruitbox-v2/master/rpi/fruitbox
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.111.133, 185.199.110.133, 185.199.109.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 7787132 (7.4M) [application/octet-stream]
Saving to: ‘fruitbox’

fruitbox                            100%[==================================================================>]   7.43M   918KB/s    in 11s

2022-07-06 01:05:06 (665 KB/s) - ‘fruitbox’ saved [7787132/7787132]

**********************************************************************
fruitbox updated!

beit-no1, 2022-07-06:
Seems I hijacked the 2 zero delay usb encoder closed issue to get your help, but thank you for the quick replies and builds.. seems to have updated.  Love the software and eager to test this version more on my JukeBox.

chundermike, 2022-07-06:
No problem.  Sorry my utilities were so bug ridden...

Now you have the latest version you should just be able to perform any future updates from the on-screen menu.

beit-no1, 2022-07-06:
Bugs are an inevitable part of the process and support takes care of them... glad to help and I look forward to that for sure... Brian

chundermike, 2022-07-13:
> Hello Mike, want to show you my fruitbox-description and my projekts in a german radio-forum. Still working on it. https://radio-bastler.de/forum/showthread.php?tid=20130 [...]

Hi again

Just read your latest article , very good again. I think this will benefit many people. One more quick tip for you...when you removed the Tint effect from the background by doing this...

tint = 255 [255 ${SPECTRUM_BAND} 0] 255 [255 ${SPECTRUM_BAND} 0] 255 [255 ${SPECTRUM_BAND} 0] 255

...you can actually make it much simpler.  The dynamic values (enclosed by square brackets) are optional, so the Tint entry can simply be...

tint = 255 255 255 255

Or even better, since the tint value defaults to 255 for each element, you can omit it altogether. :)

Cheers
Mike

saarfranzose, 2022-10-11:
Hello Mike,

 

just for information:

 

with V2.03 Keyboard works now!

 

Other problems are not solved:

 

WARNING : Couldn't backup ~/fruitbox to ~/fruitbox_backup

ln: failed to create symbolic link '/home/pi/fruitbox/rpi/fruitbox': File exists

**********************************************************************

ERROR : Couldn't create softlink to ~/fruitbox/rpi/fruitbox

 

(Fruitbox – File is missing in ~/fruitbox Directory)

 

„InstantPlay = Yes“ has no effect

 

Greetings

Jupp

 

https://radio-bastler.de/forum/showthread.php?tid=20130

https://radio-bastler.de/forum/showthread.php?tid=20296