chundermike/rpi-fruitbox-v2 #5: Disappearing Songs

Dancllns, 2022-07-12:
Hi there,

I’ve got Fruitbox installed and it’s amazing. The only issue that I have is that once the song list has cycled through once, a large number of songs vanish from the song lists. Id guess about half aren’t there anymore.

I’m using the Wurly version but it’s happened on all of the different skins. I’ve got around 500 songs on there. Is it a capacity issue? Using V2 by the way.

Would really appreciate any help as it’s the only glitch in an awesome piece of software.

chundermike, 2022-07-12:
Hi,

Sounds like a new one...can you give a little more information please?

When you say song list, do you mean a playlist loaded using the --load-playlist command line argument? Or do you just mean after you gave navigated through all the title strips using the left and right buttons?

Do the songs disappear completely from the title strips?

Do you have Loop mode enabled?

Do you have Attract mode enabled?

Can you regenerate the database and send me the fruitbox.log file by any chance?

Cheers
Mike

Dancllns, 2022-07-12:
Hi,

So it wasn’t on a playlist, but just the title strips would vanish.

I’ve just reinstalled and so far, the issue isn’t appearing again so crossing fingers and will flag if it happens again.

Out of interest, is there any capacity / performance impact when a certain number of songs are loaded?

Thanks for coming back Mike - really appreciated 

Dan Collins

closed 2022-07-14 by chundermike