chundermike/rpi-fruitbox-v2 #7: ERROR: Failed to get display mode

S3nt3nza, 2022-07-23:
Hi Mike and congratulations on your work.

I have a 2GB Rpi4 with a clean installation of 64-bit Raspberry Pi OS, after executing your instructions to the letter I execute the command "./fruitbox" but I get the error in question. Because?

Thanks and sorry for the bad English

chundermike, 2022-07-23:
Did you execute the command from a remote terminal?  Or not boot into the windows GUI, but boot and run from command line ?

S3nt3nza, 2022-07-24:
Oooh damn, you're right I was running the command from PuTTY. Running the command from the Raspbian terminal worked. Great Mike

Excuse the ignorance but what's the difference?

Again my compliments on your work.
Greetings

closed 2022-07-24 by S3nt3nza

chundermike, 2022-07-24:
Hi

The difference is due to the way Linux handles graphic displays. When you open a terminal on the Pi itself, you are operating within the desktop environment,  which has a graphical display already, and is available to any local application. When you open a Putty terminal, you are operating remotely, away from that desktop environment.  So when fruitbox tries to open a graphical display, it has to create a new one on the computer it's running on, but fails. You can configure Putty to support graphics (I've done this in the past bit can't remember exactly what to do. I think you need to enable X11 forwarding in Putty and install an X server on your PC . Maybe worth a Google?).

That's for Pi4. For Pi3 and below fruitbox runs fine locally and remotely. This is because prior to Pi4, the graphics drivers were not standard Linux but specifically written for the Pi, so don't behave the same in some cases. So I assume you're running on a Pi4?

Cheers
Mike

S3nt3nza, 2022-07-24:
Thanks Mike, you were very clear.

Until next time

chundermike, 2022-10-11:
Are you running from a remote  terminal?

S3nt3nza, 2022-10-12:
Hi Mike, I solved the problem some time ago by running the command from the Raspbian terminal and not from Putty (in a windows environment).

Thanks again