chundermike/rpi-fruitbox-v2 #8: Fruitbox gets slower and then freezes and crashes

werrie90, 2022-07-24:
Hi Mike,

Thanks for the amazing piece of software you have provided us. 

I have a raspberry pi 3 b and running version v2.03

I'm using the indieglo skin

Have about 2000 odd songs on it. 

I noticed that the titles trip auto turn time gets slower after a while. Then the titles will get a weird color change and the. Just errors. After that it crashes.

When I load it up it runs smoothly again until about 3 or 4 hours later when it crashes. 

Anything I can check out for? 

The cpu and memory are at about 20% range. This happens when nothing is playing.

chundermike, 2022-07-25:
Hi,

Do you get any errors on the terminal? (Specifically GL error 505)?  I will try and reproduce your issue.  Thanks for reporting the problem.

Cheers
Mike

werrie90, 2022-07-31:
Hi Mike, sorry have not been able to touch my computer this week. will run it today and get the errors

werrie90, 2022-07-31:
Hi, The error I am getting is g1GetError 0x505

werrie90, 2022-07-31:
[image]

So it runs normally, then gets slower and slower and usually the colors on the title strips change. Then it crashes and throughs those errors. 

The errors just stop after some time and it goes back to prompt. 

Will be doing a fresh install today to test.

chundermike, 2022-07-31:
That error means the GPU has run out of memory. You can increase it by running sudo raspi-config, then changing the memory split in the advanced options. I recommend at least 256, but you may need to increase it further.

Mike

werrie90, 2022-07-31:
Okay great thanks. 

I did have it on 256mb, have increased it now. 

But what I find weird is that it works fine for a few hours. Then gradually goes slower until the crash. 
When i first start, the memory usage and everything looks great. 

Will advise, have just increased and leaving it to run again.

chundermike, 2022-07-31:
Ok thanks. Maybe I have a memory leak. I'm on holiday next week, but will try and reproduce the issue when I get back.

Cheers
Mike

werrie90, 2022-07-31:
No worries, hope you have a great holiday. 

This can wait

beit-no1, 2022-07-31:
Hi there,
I took the attached screenshot yesterday and was going to post.  I would say there is a memory leak.  The old version I could leave my jukebox running for weeks, but this version I find it crashed easily within a day.  I also have 256 set for memory.  I could start timing it, but will wait for you to check..
[image]

chundermike, 2022-07-31:
> Hi there, I took the attached screenshot yesterday and was going to post. I would say there is a memory leak. The old version I could leave my jukebox running for weeks, but this version I find it [...]

Thanks for the feedback. I will definitely focus on assuming a memory leak then.

Cheers
Mike

werrie90, 2022-07-31:
[image]
[image]

![20220731_192809](https://user-images.githubusercontent.com/9069944/182038518-7112e

https://user-images.githubusercontent.com/9069944/182038519-fc7703bd-0255-4266-bbd6-bac25f42afc7.mp4

ebc-a6ec-4ce4-b276-967a8c735342.jpg)

Hi Mike,

Just some more info. Had increased the mem to 512. It ran a lot longer. But currently going crazy. 

Have attached some items for you. Hope it helps.

beit-no1, 2022-08-26:
Any ETA on a version update to help with the suspected memory leak.. rebooting daily now to keep box stable

beit-no1, 2022-09-27:
@chundermike Just wondering if you have any timeline on a version to stop the memory leak and crashing... I'm considering going back to the previous version as my JukeBox constantly crashes right now on fruitbox-v2.

chundermike, 2022-09-28:
Hi,

Difficult to pin an exact timeline down, but I'm looking at the issue now and it's my top fruitbox priority.

Cheers
Mike

beit-no1, 2022-09-28:
Thanks Mike... that is good enough for me to wait.

Links:
------
[1] 
https://github.com/chundermike/rpi-fruitbox-v2/issues/8#issuecomment-1260476074
[2] 
https://github.com/

chundermike, 2022-10-11:
Hi

Sorry not had the chance to look at it yet; have been on holiday. Will take a look at fixing it any day soon.

MickeyKnox, 2022-11-03:
Mike, have you had a chance to check this yet? Was there a memory leak?

chundermike, 2022-11-04:
Hi, sorry I've been snowed under with other stuff, but will try and take a look very soon

Cheers
Mike

MickeyKnox, 2022-11-05:
i only put in 100 songs and didn't play anything. I then just let it sit on the indiglo theme for about 4 hours and it froze up. not sure if its related to the software or not. 
I think it could be related to the number of pages it has to generate for the number of songs. Would be interested in knowing if there is a limit to the number of songs to add to the music. the first guy had 2000 songs. maybe if you put in 500 or less it won't crash.

MickeyKnox, 2022-11-06:
I'll throw a hundred bucks at you to get this fixed. (Sorry not rich) :-)

chundermike, 2022-11-06:
Thanks for the info, I'll try and reproduce the bug using that skin and number of songs. I'll look into it as a priority.

beit-no1, 2022-11-06:
Happy to test as my new installation with v2 crashes each day if just left unused, so it certainly appears to be a memory leak... I can give more details on the screen and number of songs, but at this time it's nothing unreasonable.

MickeyKnox, 2022-11-07:
> 

but are you using indiglo? I'm wondering if its the skin. I haven't used any other.

jaythom1, 2022-11-10:
I'll submit some un-released skins once you fix it Mike. LOL

I'm curious what part of the skin is contributing the most to the slow down and crash. Is it the sequence of jpgs for the record spin, the multiple different colored title strips or a combination of the two? For what it's worth, my pi 400 will run for days and days with one of these skins selected with no issues. Have we hit a memory ceiling with the pi3s?

chundermike, 2022-11-10:
Hi Jay

Sorry I've been a bit slow the last couple of months...recently changed my job and house still being renovated so have been a bit distracted. I've put some Debug into my code now so will begin looking into the possible memory leak. If it is a particularly skin causing the issue then it still is the fault of fruitbox not the skin, as fruitbox should be able to handle (or gracefully moan at) any skin config.  I'll keep you posted. On a different note, I have a nice video clip of a jukebox record loading (like Wurly but a different machine), if you fancy working your magic on that?

jaythom1, 2022-11-11:
Hi Mike. Yeah, I'd love to have a go at your record loading video. Maybe you can throw it in a share and send me a link?

chundermike, 2022-11-11:
Great, will do

chundermike, 2022-11-15:
Issue now found and fixed. Will get a release out in the next daybor two.

chundermike, 2022-11-15:
v2.04 now released.

chundermike, 2022-11-15:
> Thanks Mike! Looking forward to finishing my Lower Decks theme with improved spectrum analyzer color settings!

Great. As soon as you're happy to share your skins, send them to me and I'll stick them on Github :)

beit-no1, 2022-11-15:
Hi Mike.. been waiting for this memory leak a while, so very excited to try as my installation has been crashing daily.. 

Went through the menu right now with Tab and can get the v2.04 to download, but there is no restart menu and if I power off and then power on or do sudo reboot then it still load v2.03.   Is there another recommended method to do this or is it a bug as I don't want to ruin my installation.

Thanks for this.

chundermike, 2022-11-16:
That's weird, it should download the new fruitbox executable into rpi/fruitbox or rpi4/fruitbox (depending on which pi you have), and then in the fruitbox directory there should be a soft link called fruitbox to it. Try going into the rpi or rpi4 directory itself and typing. /fruitbox --version, to see if it's 2.04. If so, then the softlink in the fruitbox directory is wrong

beit-no1, 2022-11-16:
I checked /home/pi/fruitbox/rpi   3 files
fruitbox 7,792,252 with todays date/time
fruitbox_2.03 7,792,252 with todays date/time
fruitbox_prev 7,769,360 with 5/30/2022

pi@raspberrypi:~/fruitbox/rpi $ ./fruitbox --version
2.04

pi@raspberrypi:~/fruitbox/rpi $ ./fruitbox_prev --version
2.00

pi@raspberrypi:~/fruitbox/rpi $ ./fruitbox_2.03 --version
2.04

chundermike, 2022-11-16:
That's downloaded it OK. Have you manually changed the softlink?

Go into the fruitbox directory and type

ls -al

...the fruitbox entry should point to rpi/fruitbox.  If it doesn't,  do the following...

cd /home/pi/fruitbox
rm ./fruitbox
ln -s rpi/fruitbox ./fruitbox

Then run fruitbox in the current directory (by typing ./fruitbox --version) it should be 2.04

Mike

2.04

2.00

2.04

beit-no1, 2022-11-16:
Was definitely pointing to a v2.03.. not sure if from the last upgrade, but I recall having some issues with the move to v2.03.  Your instructions seem to have worked

[image]

closed 2022-11-18 by chundermike