chundermike/rpi-fruitbox-v2 #9: SB-M100 Touch Skin Error

jstoker24, 2022-09-03:
Thanks for making the project. I'm in the middle of making a case and just getting the pi and fruitbox configured and ready to fit and use 

My display is an adafruit 7" display with mini driver, and it all works part from one par which is the to SB-M100 Touch Skin  when I load I see the fruitbox startup logo then I get the error  as shown below 

[image]

I have also added I'm my fruitbox.log  below as well.

fruitbox version 2.03 (Jul  5 2022, 23:07:24) by Mike Kingsley
A customisable MP3/OGV Jukebox for the Raspberry Pi.
Free for non-commercial use
Raspberry Pi Hardware version 4 detected
Detecting display...1024 x 600 (60Hz)
Loading initialisation file 'fruitbox.ini'...

  [general]
  MusicPath = ../Music
  MaxPlayQueueLength = 5
  LoopPlayQueue = No
  IgnoreArtistPrefixThe = Yes
  SongHistoryDepth = 20
  SongHistoryIncludeUserSelection = No
  SongVolume = 50
  SongVolumeStep = 10
  ReplayGain = None
  FreePlay = No
  CreditsPerCoin1 = 1
  CreditsPerCoin2 = 2
  CreditsPerCoin3 = 5
  CreditsPerCoin4 = 10
  CreditsPerCoin5 = 25
  CreditsPerCoin6 = 100
  CreditsPerSong = 1
  MenuColour = 255 255 255
  MenuBackgroundColour = 32 32 32 200

  [attract mode]
  Enable = No
  Choice = Random
  Interval = 1.0 Minute
  MaxVolume = 50
  FadeIn = 5.0 Seconds
  FadeOut = 10.0 Seconds
  FadeInPosition = 60.0 Seconds
  FadeOutPosition = 70.0 Seconds
  SkipLoad = Yes

  [skins]
  File = skins/SB-M100/fruitbox_touch.cfg

  [buttons]
  Quit = KEY_ESC
  Coin1 = KEY_F1
  Coin2 = KEY_F2
  Coin3 = KEY_F3
  Coin4 = KEY_F4
  Coin5 = KEY_F5
  Coin6 = KEY_F6
  VolUp = KEY_EQUAL
  VolDown = KEY_MINUS
  Random = KEY_SPACE
  Select = KEY_ENTER
  Skip = KEY_BACKSPACE
  Pause = KEY_SEMICOLON
  Up = KEY_UP
  Down = KEY_DOWN
  Left = KEY_LEFT
  Right = KEY_RIGHT
  LeftScreen = KEY_LEFTBRACE
  RightScreen = KEY_RIGHTBRACE
  LeftSort = KEY_COMMA
  RightSort = KEY_DOT
  AttractMode = KEY_APOSTROPHE
  LoopQueue = KEY_F11
  FreePlay = KEY_F12
  ClearQueue = KEY_DELETE
  Mute = KEY_LEFTCTRL
  PowerOff = KEY_SYSRQ
  Select0 = KEY_0
  Select1 = KEY_1
  Select2 = KEY_2
  Select3 = KEY_3
  Select4 = KEY_4
  Select5 = KEY_5
  Select6 = KEY_6
  Select7 = KEY_7
  Select8 = KEY_8
  Select9 = KEY_9
  SelectA = KEY_A
  SelectB = KEY_B
  SelectC = KEY_C
  SelectD = KEY_D
  SelectE = KEY_E
  SelectF = KEY_F
  SelectG = KEY_G
  SelectH = KEY_H
  SelectI = KEY_I
  SelectJ = KEY_J
  SelectK = KEY_K
  SelectL = KEY_L
  SelectM = KEY_M
  SelectN = KEY_N
  SelectO = KEY_O
  SelectP = KEY_P
  SelectQ = KEY_Q
  SelectR = KEY_R
  SelectS = KEY_S
  SelectT = KEY_T
  SelectU = KEY_U
  SelectV = KEY_V
  SelectW = KEY_W
  SelectX = KEY_X
  SelectY = KEY_Y
  SelectZ = KEY_Z
  Flag1 = KEY_F7
  Flag2 = KEY_F8
  Flag3 = KEY_F9
  Flag4 = KEY_F10
  Menu = KEY_TAB
  Search = KEY_SLASH
  Mouse = BTN_LEFT
  Screenshot = KEY_RIGHTCTRL

  [touch screen]
  Enable = Yes
  TopLeft = 27 -52
  BottomRight = 4093 4108

  [mouse]
  Enable = No
  Size = 0 0
  Angle = 0
  Tint = 255 255 255 255
  HorizontalFlip = No
  VerticalFlip = No
  HotSpot = 0 0
  Sensitivity = 1.0
  HideAfter = 5.0 Seconds

  [joystick]
  Enable = No
  DisableTitleStripFlip = No
  Size = 0 0
  Angle = 0
  Tint = 255 255 255 255
  HorizontalFlip = No
  VerticalFlip = No
  Offset = 0 0

...initialisation file closed.

Detecting input devices...
  Opened "DELL Dell USB Entry Keyboard" (/dev/input/event0)
  ...keyboard detected
  Opened "Dell Dell Universal Receiver" (/dev/input/event1)
  ...keyboard detected
  ...mouse detected
  Opened "Dell Dell Universal Receiver Mouse" (/dev/input/event2)
  Opened "Dell Dell Universal Receiver Consumer Control" (/dev/input/event3)
  Opened "Dell Dell Universal Receiver System Control" (/dev/input/event4)
  ...touchscreen detected
  Opened "Microchip Technology Inc. AR1100 HID-MOUSE" (/dev/input/event5)
  Opened "vc4" (/dev/input/event6)
  Opened "vc4" (/dev/input/event7)
  Opened GPIO

Reading skin configuration file 'skins/SB-M100/fruitbox_touch.cfg'...

  [general]
  SkinName = SB-M100
  SkinDescription = by Keefypee
  SkinSize = 1920 1080
  SortSongsBy = Title
  SortSongsBy = Artist
  SelectButtons = ABCD
  SelectButtons = 12345678
  SelectButtonsSequence = RowCol
  AutoSelect = No
  SelectTimeout = 4.0 Seconds
  SelectHoldTimeout = 2.0 Seconds
  SongsPerTitleStrip = 2
  TitleStripAutoTurnTime = 1.0 Minute
  TitleStripSize = 360 100
  TitleStripBackground = skins/SB-M100/page_bkgnd.png 
  TitleStripMoveStyle = 7 5 Yes
  PairSongs = Dual
  ArtistDescription = ${SONG_ARTIST} / ${PAIRED_SONG_ARTIST}
  SongLoadTime = 447.0 Ticks
  SongUnloadTime = 300.0 Ticks
  ArtistText = 1 0 0 0 255 centre Yes -1 No 0 0 260
  SongText = 1 200 0 0 255 centre Yes -1 No 0 0 350

  [status]
  Contents = skins/SB-M100/coin_insert_sound.txt
  Sound = skins/SB-M100/coin-insert.wav 90 No

  [status]
  Contents = skins/SB-M100/select_song_sound.txt
  Sound = skins/SB-M100/JBBPress.wav 100 No

  [status]
  Contents = skins/SB-M100/load_song_sound.txt
  Sound = skins/SB-M100/mech-only2.wav 90 No

  [status]
  Contents = skins/SB-M100/unload_song_sound.txt
  Sound = skins/SB-M100/mech-only-rev.wav 90 No

  [status]
  Contents = skins/SB-M100/page_move_sound.txt
  Sound = skins/SB-M100/JBPTurn.wav 80 No

  [status]
  Contents = skins/SB-M100/invalid_song_sound.txt
  Sound = skins/SB-M100/invalid.wav 100 No

  [fonts]
  File = skins/SB-M100/courierb.ttf 25
  File = skins/SB-M100/dvsansb.ttf 19
  File = skins/SB-M100/hatten.ttf 50
  File = skins/SB-M100/courier.ttf 20

  [bitmap]
  File = skins/SB-M100/foreground_touch.bmp
  Position = 0 0
  Size = 1920 1080

  [title strip]
  Position = 55 112
  Angle = 359
  MoveDelay = 10
  MoveReverse = No

  [title strip]
  Position = 55 230
  Angle = 0
  MoveDelay = 4
  MoveReverse = No

  [title strip]
  Position = 55 348
  Angle = 0
  MoveDelay = 5
  MoveReverse = No

  [title strip]
  Position = 55 465
  Angle = 0
  MoveDelay = 9
  MoveReverse = No

  [title strip]
  Position = 555 112
  Angle = 0
  MoveDelay = 7
  MoveReverse = Yes

  [title strip]
  Position = 555 230
  Angle = 0
  MoveDelay = 1
  MoveReverse = Yes

  [title strip]
  Position = 555 348
  Angle = 1
  MoveDelay = 6
  MoveReverse = Yes

  [title strip]
  Position = 555 465
  Angle = 0
  MoveDelay = 10
  MoveReverse = Yes

  [title strip]
  Position = 1000 112
  Angle = 0
  MoveDelay = 2
  MoveReverse = No

  [title strip]
  Position = 1000 230
  Angle = 359
  MoveDelay = 3
  MoveReverse = No

  [title strip]
  Position = 1000 348
  Angle = 0
  MoveDelay = 12
  MoveReverse = No

  [title strip]
  Position = 1000 465
  Angle = 0
  MoveDelay = 7
  MoveReverse = No

  [title strip]
  Position = 1505 112
  Angle = 0
  MoveDelay = 0
  MoveReverse = Yes

  [title strip]
  Position = 1505 230
  Angle = 0
  MoveDelay = 3
  MoveReverse = Yes

  [title strip]
  Position = 1505 348
  Angle = 1
  MoveDelay = 6
  MoveReverse = Yes

  [title strip]
  Position = 1505 465
  Angle = 0
  MoveDelay = 9
  MoveReverse = Yes

  [touch areas]
WARNING: Unrecognised section (line 159)

  [status]
  Position = 116 750
  Size = 314 222
  Text = 2 255 0 0 255 centre No 1 No 0 0 300
  Contents = skins/SB-M100/now_playing.txt
  TimerTickPeriod = 10.0 Ticks
  LineSpacing = 1

  [status]
  Position = 1500 812
  Size = 310 132
  Text = 2 255 0 0 255 centre No -1 No 0 0 300
  Contents = skins/SB-M100/coming_up.txt
  LineSpacing = 1

  [status]
  Position = 1480 746
  Size = 190 160
  Text = 3 255 0 0 255 centre No 0 No 0 0 160
  Contents = skins/SB-M100/selection.txt

  [status]
  Position = 1660 746
  Size = 190 160
  Text = 3 255 0 0 255 centre No 0 No 0 0 160
  Contents = skins/SB-M100/credits.txt
  TimerTickPeriod = 150.0 Ticks
Closed skin file.

Initialising audio ...

Checking music database ../Music/fruitbox.db
  Reading database file '../Music/fruitbox.db'
  3 songs found
  Sorting songs by Title ...
  Sorting songs by Artist ...
Building song Title Strips...
ERROR: Couldn't load bitmap 'skins/SB-M100/foreground_touch.bmp'

any help on this would be brilliant as it will then allow me to progress with the project

chundermike, 2022-09-03:
Hi,

It appears that there is a space in the file named foreground_touch.bmp. this is wrong and I'll correct it ASAP. In the meantime, you could rename the bmp to remove the space.

Cheers
Mike

[SB-M100 Touch Skin]<https://user-images.githubusercontent.com/111202776/188287223-16b73dcf-0d63-4031-b301-ae113565a272.png>

chundermike, 2022-09-04:
Also I've just realised the touch version of this skin is not really finished (and I've not tested it). You'll need to add a bunch of button graphics to the foreground image, and a [touch buttons] section and optionally a [touch song] section to the fruitbox.cfg file.  I've not got enough free time atm to do this I'm afraid though (I'm putting any spare time I have for fruitbox into fixing issues in the current release).

Cheers
Mike

closed 2022-09-04 by chundermike