chundermike/rpi-fruitbox-v2 #10: filter by

vigorator, 2022-09-27:
Sorry maybe there is a way to configure it but I couldn't find it...

I have around 1000 mp3 and it is hard to skip between all the pages.
Is there a way to make filters (of generes probably the easier)
so show me:
 all music
 pop music
 disco music

and depending on the situation (is it a party at home, bedtime, day) just apply filters by using one button?

randomize of mp3 should respect filter by setting as well.

chundermike, 2022-09-27:
Hi,

Not directly, but if you hit the TAB button on the keyboard (MENU_SEARCH, which can be re-mapped to anything, like the other buttons can), then a search menu will appear on screen. In the Genre field, just enter the genre string you wish to filter on, exit the menu, and only those songs with that genre will be displayed.

Mike

vigorator, 2022-09-27:
On my arcade buttons, I'm not usually going to MENU_SEARCH,
Maybe it is a good idea to make an option to configure it?
instead of skip pages, make filter_by?

Thanks for the explanation.
Soon l'Il test it.

chundermike, 2022-09-27:
Hi,

Yes perhaps that is a good idea...have one or more dedicated buttons to filter songs, in which the filter criterion is set in the fruitbox ini file...

...I'll give that some thought.

Mike

vigorator, 2022-09-27:
Thanks,
BTW this is cool software. Installed it a week ago but I'm still impressed.