chundermike/rpi-fruitbox-v2 #11: Touchscreen Calibration

jzrlost, 2022-10-03:
I have a pi 4 and a 21.5" touchscreen.  The touchscreen works fine on the pi and everything is calibrated.  I setup fruitbox-v2 to use the trouchscreen, but the "touch" on the screen is not where the buttons are located.  If I touch up by the songs A4-A5 buttons in the song title area I see the C and D buttons push.  Not sure if there a better calibration I can run in the fruitbox-v2 program or if there is anything else I can do fix the issue.  I am testing the SB-M100 touchscreen skin.  I tested the TouchOne skin and it's off also.

chundermike, 2022-11-04:
Can you post your fruitbox.ini file, and also the output of an evtest run (where you select the touch device and press top left and bottom right of the screen)?

Thanks

jzrlost, 2022-11-07:
Here is the fruitbox.ini:

# fruitbox version 2.03 global config file

[general]
  MusicPath = /music
  CustomMP3Tag =
  MaxPlayQueueLength = 100
  LoopPlayQueue = No
  IgnoreArtistPrefixThe = Yes
  SongHistoryDepth = 20
  SongHistoryIncludeUserSelection = No
  SongVolume = 40
  SongVolumeStep = 10
  ReplayGain = None
  FreePlay = Yes
  CreditsPerCoin1 = 1
  CreditsPerCoin2 = 2
  CreditsPerCoin3 = 5
  CreditsPerCoin4 = 10
  CreditsPerCoin5 = 25
  CreditsPerCoin6 = 100
  CreditsPerSong = 1
  MenuColour = 255 255 255
  MenuBackgroundColour = 32 32 32 200

[attract mode]
  Enable = Yes
  Choice = Random
  Interval = 1.0 Minute
  MaxVolume = 50
  FadeIn = 5.0 Seconds
  FadeOut = 10.0 Seconds
  FadeInPosition = 60.0 Seconds
  FadeOutPosition = 70.0 Seconds
  SkipLoad = Yes

[skins]
  File = skins/SB-M100/fruitbox_touch.cfg

[buttons]
  Quit = KEY_ESC
  Coin1 = KEY_F1
  Coin2 = KEY_F2
  Coin3 = KEY_F3
  Coin4 = KEY_F4
  Coin5 = KEY_F5
  Coin6 = KEY_F6
  VolUp = KEY_EQUAL
  VolDown = KEY_MINUS
  Random = KEY_SPACE
  Select = KEY_ENTER
  Skip = KEY_BACKSPACE
  Pause = KEY_SEMICOLON
  Up = KEY_UP
  Down = KEY_DOWN
  Left = KEY_LEFT
  Right = KEY_RIGHT
  LeftScreen = KEY_LEFTBRACE
  RightScreen = KEY_RIGHTBRACE
  LeftSort = KEY_COMMA
  RightSort = KEY_DOT
  AttractMode = KEY_APOSTROPHE
  LoopQueue = KEY_F11
  FreePlay = KEY_F12
  ClearQueue = KEY_DELETE
  Mute = KEY_LEFTCTRL
  PowerOff = KEY_SYSRQ
  Select0 = KEY_0
  Select1 = KEY_1
  Select2 = KEY_2
  Select3 = KEY_3
  Select4 = KEY_4
  Select5 = KEY_5
  Select6 = KEY_6
  Select7 = KEY_7
  Select8 = KEY_8
  Select9 = KEY_9
  SelectA = KEY_A
  SelectB = KEY_B
  SelectC = KEY_C
  SelectD = KEY_D
  SelectE = KEY_E
  SelectF = KEY_F
  SelectG = KEY_G
  SelectH = KEY_H
  SelectI = KEY_I
  SelectJ = KEY_J
  SelectK = KEY_K
  SelectL = KEY_L
  SelectM = KEY_M
  SelectN = KEY_N
  SelectO = KEY_O
  SelectP = KEY_P
  SelectQ = KEY_Q
  SelectR = KEY_R
  SelectS = KEY_S
  SelectT = KEY_T
  SelectU = KEY_U
  SelectV = KEY_V
  SelectW = KEY_W
  SelectX = KEY_X
  SelectY = KEY_Y
  SelectZ = KEY_Z
  Flag1 = KEY_F7
  Flag2 = KEY_F8
  Flag3 = KEY_F9
  Flag4 = KEY_F10
  Menu = KEY_TAB
  Search = KEY_SLASH
  Mouse = BTN_LEFT
  Screenshot = KEY_RIGHTCTRL

[touch screen]
  Enable = Yes
  TopLeft = 2633 2786
  BottomRight = 13809 13756

[mouse]
  Enable = Yes
  Bitmap =
  Size = 44 67
  Angle = 0
  Tint = 255 255 255 255
  HorizontalFlip = No
  VerticalFlip = No
  HotSpot = 2 2
  Sensitivity = 1.0
  HideAfter = 5.0 Seconds

[joystick]
  Enable = No
  DisableTitleStripFlip = No
  Bitmap =
  Size = 0 0
  Angle = 0
  Tint = 255 255 255 255
  HorizontalFlip = No
  VerticalFlip = No
  Offset = 0 0

How do I run the evtest stand-alone.  The only way I know how to get it run is to remove the "TopLeft and BottomRight" lines in the fruitbox.ini.  I did get the output from the touchscreen program in Raspbian.  Here is that output:

Warning: multiple calibratable devices found, calibrating last one (Weida Hi-Tech                CoolTouchR System           )
        use --device to select another one.
Calibrating standard Xorg driver "Weida Hi-Tech                CoolTouchR System           "
        current calibration values: min_x=0, max_x=65535 and min_y=0, max_y=65535
        If these values are estimated wrong, either supply it manually with the --precalib option, or run the 'get_precalib.sh' script to automatically get it (through HAL).
        --> Making the calibration permanent <--
  copy the snippet below into '/etc/X11/xorg.conf.d/99-calibration.conf' (/usr/share/X11/xorg.conf.d/ in some distro's)
Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "Weida Hi-Tech                CoolTouchR System           "
        Option  "MinX"  "-134"
        Option  "MaxX"  "65788"
        Option  "MinY"  "126"
        Option  "MaxY"  "65621"
        Option  "SwapXY"        "0" # unless it was already set to 1
        Option  "InvertX"       "0"  # unless it was already set
        Option  "InvertY"       "0"  # unless it was already set
EndSection

The monitor is a ViewSonic TD2230.

jzrlost, 2022-12-06:
Mike,
  Any ideas on this issue?  I can look at sending you a touchscreen for testing if that would make it easier to figure out the issue.

chundermike, 2022-12-07:
Hi,

The loan of a touchscreen would be the best option, but did you try evtest? That would be less.hassle for you.

Mike

jzrlost, 2022-12-07:
I ran both "evtest" and started fruitbox screen configuration at the same time.  Here is the evtest output:

jukebox@raspberrypi:~ $ evtest
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:      Logitech USB Optical Mouse
/dev/input/event1:      Dell Dell USB Keyboard
/dev/input/event2:      Weida Hi-Tech                CoolTouchR System            Mouse
/dev/input/event3:      Weida Hi-Tech                CoolTouchR System
/dev/input/event4:      vc4
/dev/input/event5:      vc4
Select the device event number [0-5]: 3
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x2575 product 0x401 version 0x110
Input device name: "Weida Hi-Tech                CoolTouchR System           "
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value  23361
      Min        0
      Max    32767
      Resolution      69
    Event code 1 (ABS_Y)
      Value  15459
      Min        0
      Max    32767
      Resolution     122
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        9
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max    32767
      Resolution      69
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max    32767
      Resolution     122
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
  Event type 4 (EV_MSC)
    Event code 5 (MSC_TIMESTAMP)
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
Event: time 1670440060.003598, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 4
Event: time 1670440060.003598, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 4064
Event: time 1670440060.003598, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 4035
Event: time 1670440060.003598, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1670440060.003598, type 3 (EV_ABS), code 0 (ABS_X), value 4064
Event: time 1670440060.003598, type 3 (EV_ABS), code 1 (ABS_Y), value 4035
Event: time 1670440060.003598, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 0
Event: time 1670440060.003598, -------------- SYN_REPORT ------------
Event: time 1670440060.019588, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 4600
Event: time 1670440060.019588, -------------- SYN_REPORT ------------
Event: time 1670440060.035623, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 9000
Event: time 1670440060.035623, -------------- SYN_REPORT ------------
Event: time 1670440060.051560, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 13400
Event: time 1670440060.051560, -------------- SYN_REPORT ------------
Event: time 1670440060.067563, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 18800
Event: time 1670440060.067563, -------------- SYN_REPORT ------------
[... 77 lines cut ...]
Event: time 1670440062.387807, type 3 (EV_ABS), code 0 (ABS_X), value 28764
Event: time 1670440062.387807, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 84300
Event: time 1670440062.387807, -------------- SYN_REPORT ------------
Event: time 1670440062.403803, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 28763
Event: time 1670440062.403803, type 3 (EV_ABS), code 0 (ABS_X), value 28763
Event: time 1670440062.403803, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 93800
Event: time 1670440062.403803, -------------- SYN_REPORT ------------
Event: time 1670440062.419863, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 28765
Event: time 1670440062.419863, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 28761
Event: time 1670440062.419863, type 3 (EV_ABS), code 0 (ABS_X), value 28765
Event: time 1670440062.419863, type 3 (EV_ABS), code 1 (ABS_Y), value 28761
Event: time 1670440062.419863, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 103300
Event: time 1670440062.419863, -------------- SYN_REPORT ------------
Event: time 1670440062.435778, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 112800
Event: time 1670440062.435778, -------------- SYN_REPORT ------------
Event: time 1670440062.451776, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 122300
Event: time 1670440062.451776, -------------- SYN_REPORT ------------
Event: time 1670440062.467771, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 131800
Event: time 1670440062.467771, -------------- SYN_REPORT ------------
Event: time 1670440062.483758, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1670440062.483758, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1670440062.483758, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 141300
Event: time 1670440062.483758, -------------- SYN_REPORT ------------

Let me know if that helps.

chundermike, 2022-12-07:
Hi

That's getting more useful. Can you just run evtest on its own, select the touchscreen and press screen top left then screen bottom right, and send me the evtest output. Hopefully it should give us the touch coordinates of the two screen corners.

Thanks.

jzrlost, 2022-12-07:
jukebox@raspberrypi:~ $ evtest
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:      Logitech USB Optical Mouse
/dev/input/event1:      Dell Dell USB Keyboard
/dev/input/event2:      Weida Hi-Tech                CoolTouchR System            Mouse
/dev/input/event3:      Weida Hi-Tech                CoolTouchR System
/dev/input/event4:      vc4
/dev/input/event5:      vc4
Select the device event number [0-5]: 3
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x2575 product 0x401 version 0x110
Input device name: "Weida Hi-Tech                CoolTouchR System           "
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value  28765
      Min        0
      Max    32767
      Resolution      69
    Event code 1 (ABS_Y)
      Value  28761
      Min        0
      Max    32767
      Resolution     122
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        9
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max    32767
      Resolution      69
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max    32767
      Resolution     122
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
  Event type 4 (EV_MSC)
    Event code 5 (MSC_TIMESTAMP)
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
Event: time 1670441819.407895, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 6
Event: time 1670441819.407895, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 283
Event: time 1670441819.407895, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 266
Event: time 1670441819.407895, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1670441819.407895, type 3 (EV_ABS), code 0 (ABS_X), value 283
Event: time 1670441819.407895, type 3 (EV_ABS), code 1 (ABS_Y), value 266
Event: time 1670441819.407895, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 0
Event: time 1670441819.407895, -------------- SYN_REPORT ------------
Event: time 1670441819.423891, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 4400
Event: time 1670441819.423891, -------------- SYN_REPORT ------------
Event: time 1670441819.439881, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 8800
Event: time 1670441819.439881, -------------- SYN_REPORT ------------
Event: time 1670441819.455914, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 13100
Event: time 1670441819.455914, -------------- SYN_REPORT ------------
Event: time 1670441819.471924, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 17500
Event: time 1670441819.471924, -------------- SYN_REPORT ------------
Event: time 1670441819.487883, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 26500
Event: time 1670441819.487883, -------------- SYN_REPORT ------------
[... 175 lines cut ...]
Event: time 1670441824.192378, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 32597
Event: time 1670441824.192378, type 3 (EV_ABS), code 0 (ABS_X), value 32523
Event: time 1670441824.192378, type 3 (EV_ABS), code 1 (ABS_Y), value 32597
Event: time 1670441824.192378, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 220400
Event: time 1670441824.192378, -------------- SYN_REPORT ------------
Event: time 1670441824.208380, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 32595
Event: time 1670441824.208380, type 3 (EV_ABS), code 1 (ABS_Y), value 32595
Event: time 1670441824.208380, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 229900
Event: time 1670441824.208380, -------------- SYN_REPORT ------------
Event: time 1670441824.224383, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 32522
Event: time 1670441824.224383, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 32591
Event: time 1670441824.224383, type 3 (EV_ABS), code 0 (ABS_X), value 32522
Event: time 1670441824.224383, type 3 (EV_ABS), code 1 (ABS_Y), value 32591
Event: time 1670441824.224383, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 239400
Event: time 1670441824.224383, -------------- SYN_REPORT ------------
Event: time 1670441824.240376, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 248900
Event: time 1670441824.240376, -------------- SYN_REPORT ------------
Event: time 1670441824.256375, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 258400
Event: time 1670441824.256375, -------------- SYN_REPORT ------------
Event: time 1670441824.272340, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 267900
Event: time 1670441824.272340, -------------- SYN_REPORT ------------
Event: time 1670441824.288318, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1670441824.288318, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1670441824.288318, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 277400
Event: time 1670441824.288318, -------------- SYN_REPORT ------------

jzrlost, 2022-12-26:
Mike,
   Is there anything else you need from me to figure out this issue?  Any thoughts on why it's not using the correct resolution when it comes to the touchscreen?

thx Josh