chundermike/rpi-fruitbox-v2 #13: No Autoplay in V2?

jburg84, 2022-10-20:
Hello Mike,

first of all congratulations and thank you so much for this nice piece of work! 👍 

I'm starting my V2 project now. But there is one feature missing, that I really liked in V1.
And that's the Autoplay mode. I couldn't find it in button config any more.
Is there a workaround or was I just to blind to see it? :-D

Thanks a lot, Jens

chundermike, 2022-10-20:
Hi Jens

It's been renamed to attract mode, and enhanced. In the fruitbox.ini there is a section for it. If you don't want songs to fade in and/or out, just remove those settings completely.

Cheers
Mike

jburg84, 2022-10-21:
Hi Mike,

this is great! I'm getting more and more impressed. 👍 

Thank you so much, 
Jens

closed 2022-10-26 by chundermike