chundermike/rpi-fruitbox-v2 #14: Run on boot question

bmaselli13, 2022-10-26:
I've tried to setup the system to boot to Fruitbox. When I adjust the settings in rc.local, the system still boots to the desktop. Am I missing something?

bmaselli13, 2022-11-04:
Just bumping this

chundermike, 2022-11-04:
Ah sorry for delay...not 100% sure. As its not really a fruitbox issue maybe it's worth googling generally?

bmaselli13, 2022-11-07:
Fair enough. Was hoping others who've ran into it might chime in as well. Thanks Mike!

bmaselli13, 2022-11-10:
Worked the start on boot using autostart but now running into an issue when Fruitbox starts. I get an error with the .ini file. (See Attached)

[image]

chundermike, 2022-11-10:
The [skins] section is either empty of points to cfg files which don't exist (or their paths are wrong, most probably)

chundermike, 2022-11-10:
Not sure if it's just a formatting issue in the github messaging, but [skins] should be on a separate line to the File = line ...

[skins]
File = skins/TouchBlue/fruitbox.cfg

...and this is in the ini file, not cfg file

bmaselli13, 2022-11-10:
Sorry. First file is the .ini file. My bad on that typo.

bmaselli13, 2022-11-10:
yes, it actually reads

[skins]
  File = skins/TouchBlue/fruitbox.cfg

bmaselli13, 2022-11-10:
running from /home/bmaselli/fruitbox
`
bmaselli@raspberrypi:~/fruitbox $ ls
fruitbox      install_fruitbox      release_notes.txt  skins
fruitbox.ini  install_fruitbox_pi4  rpi                userguide.txt
fruitbox.log  README.md             rpi4
`

in skin folder

`
bmaselli@raspberrypi:~/fruitbox/skins $ ls
CDPlayer2000   JaysIndieglo     NewRetro    SB-M100    WallJuke2
Classic        JaysMoldyOldies  NumberOne   Terminal   WallSmall
JaysBigScreen  MikeTV           RadiOldie   TouchBlue  WoodysJukebox
JaysCDJukebox  Modern           readme.txt  TouchOne   Wurly
`

chundermike, 2022-11-10:
Looks OK.. . can you post fruitbox.log please?

bmaselli13, 2022-11-10:
`
fruitbox version 2.03 (Jul  5 2022, 23:07:24) by Mike Kingsley
A customisable MP3/OGV Jukebox for the Raspberry Pi.
Free for non-commercial use
Raspberry Pi Hardware version 4 detected
Detecting display...ERROR: Failed to get display mode
`

chundermike, 2022-11-10:
I thought the issue was with the skin cfg, not display mode? I'm confused now!

bmaselli13, 2022-11-10:
This all happens when it boots up. 
Currently, I am using autostart to launch fruitbox
( /etc/xdg/lxsession/LXDE-pi/autostart)

This is the last line in that file
`@lxterminal -e /home/bmaselli/fruitbox/./fruitbox`

chundermike, 2022-11-10:
I'm getting a bit out of my depth now, as I've never seen these type of issues before. Dare I say it, maybe worth starting from scratch? Sorry, not a lot of help I know.

bmaselli13, 2022-11-10:
It's ok. II appreciate any guidance you've already given. HUGE fan of the product. I'll continue on and see. Thanks!

bmaselli13, 2022-11-10:
> I thought the issue was with the skin cfg, not display mode? I'm confused now!

The display mode isn't affecting anything. Tthis is all about the cfg file issue.

chundermike, 2022-11-10:
I will keep.thinking about the issue you have...if I  manage to reproduce it I'll let you know.

bmaselli13, 2022-11-11:
Thanks. Appreciate it.