chundermike/rpi-fruitbox-v2 #16: "no music found"

BobbyBoyGaming, 2022-11-10:
the config.ini of fruitbox says that the music will be found in "../music"

Shouldnt it say "./music" with only one period instead? I try to change the text in the config.ini file but it says i dont have permission. 

And when I tried to change the permissions within RaspiOS GUI I get that I dont have permission to do that either lol...

BobbyBoyGaming, 2022-11-10:
If I launch the program as follows
"./fruitbox --music-path ./music"
it also gives me error "no music found" even though I have the mp3s in ./music

BobbyBoyGaming, 2022-11-10:
it is loading the music now, it is "./Music" not "./music" i forget that linux is case sensitive

BobbyBoyGaming, 2022-11-10:
the default fruitbox.ini is autofilled with "../Music" which may not be the same as "./Music" and I cannot alter that file, it wont let me make any changes even though I am the owner of my computer or whatever, I dont know how to superuser myself forever at all times. 

I ran sudo su and it works for the CLI, but I dont know how to change that line of text in frutbox.ini via the CLI alone, I need to do it using the OS GUI, but I cant seem to get superuser authorization in the OS or something, every time I try to change the text in "fruitbox.ini" via the OS GUI I get an error that I dont have permission/authorization. 

I also went into the directory of the default skin and looked for any kind of config.ini to see if it had a different music path, and I was only able to spot a file called "fruitbox.cfg" and I opened it in a text editor and I could not find any line of text that has anything to do with music paths. It is not clear which configuration file the --help command is referring to under the --music-path <filepath> line, wherein it says that there will be a way to configure the music path within the skin's folder

chundermike, 2022-11-10:
You can specify MusicPath values in the skin cfg files too. It will override the MusicPath value in the ini. It should explain this in the userguide. No idea why all your permissions are screwed up. If you drop to  command line, goto the directory where the ini file is  and type

sudo chmod 777 fruitbox.ini

...rhen non superusers should be able to edit the ini file

BobbyBoyGaming, 2022-11-10:
Perfect thanks.