chundermike/rpi-fruitbox-v2 #17: install it on Ubuntu 64bit?

vigorator, 2022-11-10:
Hi, Is there a way to install it on a Ubuntu 22.04.1 LTS 64 bit?
I can see in a documentation that it is only for raspberry but I need to run it somehow on a desktop ;(

I got this: ./install_fruitbox_pi4: cannot execute binary file: Exec format error
after some fixes I got Could not open '/lib/ld-linux-aarch64.so.1': No such file or directory

Not sure is it worth to fix it and basically goes this route if it won't work.

Kris

chundermike, 2022-11-10:
Soon I'll release the source code so you can compile it for any Linux based OS. They're be no GPIO support though as this is Raspberry Pi hardware specific.

vigorator, 2022-11-10:
That's great,
sorry for asking but you think is it week, month or more?
Just wondering should I install raspberry pi Desktop version and try with that?

chundermike, 2022-11-10:
Probably won't be for a few weeks yet.

vigorator, 2022-11-10:
Cool, thanks for quick response.
I will wait then!

BobbyBoyGaming, 2022-11-10:
Does the current RaspberryPi OS version support a GPIO Soundcard? I spent top dollar on a thing called "Hi-fi berry" and I may have been ripped off. 
I am afraid this Hi-fi Berry soundcard may not be compatible with anything at all, and is just basically a paperweight.

It is supposed to be for RPi and RaspiOS is only for RPi, so there may be a chance that this Hi-fi Berry could work with Fruitbox under the RaspberryPi OS ?

chundermike, 2022-11-10:
Not sure, but I have a HiFiBerry sound card (Amp+). Haven't used it for a while but it worked a treat. I'd be surprised if it isn't supported anymore, as they are very popular AFAIK.

Der-Festus, 2022-11-11:
Hi together
I have an RPI 3b with Fruitbox V1, and the Soundcard hifiberry dacplus is working fine.. 
 (You must disable the standard soundcard, so then it works fine with lot of better sound)

in my case with Fruitbox Vers. V2 no sound will come out  :(
I did not find a way to get it work till now...

 Greetings

chundermike, 2022-11-18:
Source code is now released, so you can have a go at building it for Ubuntu. I would follow the instructions in build/buid.txt for the Pi4 version, but remove -D_RPI from the make file because there will be no GPIO to support on your PC.  You may also need to build the libmpg123 library for Ubuntu (or at least hardware agnostic target) rather than targetting ARM; but you will need to visit the libmpg123 website itself for help on that (it'll be a slightly different configure command).  Good luck!

closed 2022-11-18 by chundermike

vigorator, 2022-11-18:
Thanks a lot!
I'll give it a try during weekend!