chundermike/rpi-fruitbox-v2 #18: Make my own skin

BobbyBoyGaming, 2022-11-10:
Hello Mike, in your opinion is making a skin an "advanced" task?
I want to make a skin that shows as many songs as possible on the screen, basically just like a list really. Nothing fancy, a list of songs with the ID3 Tags. Kind of like when you select "View Details" in the Taskbar on Windows 10.

And in terms of "eye Candy" I wanted to just have a second monitor just display the "Attract Mode"-type of stuff or the "spinning disk" animation.
But for me the important part is to have the process of search for music as high-yield/efficient/fast as possible, which can be achieved by rapid page-flipping without any fancy animations, and/or making use of the screen space as much as possible by cramming as many songs on the screen as possible.

I want to have a lot of music on it so the current skins are a little bit too unresponsive and slow for my liking.
In terms of HID input devices I am limited to a numberpad and two horizontal arrows for page flipping/scrolling. (unless I were to completely revamp the cabinet).
I also dont  have a keyboard or A-D letter keypad for search for employing alphanumerical codes, which the current skins seem to rely on. 

So I am interested in using purely numerical codes or simply a superfast page scrolling method (page up/page down).

I can draw up or even create a skin on corel draw or whatever, but I wouldnt really know what to do afterwards, i.e. I dont really know how I would implement it into the fruitbox. do you know if I can ask one of the previous skin creators about this? or work with someone's guidance to get my skin functional? 
Also, if it turns out really nice I would of course share it and you could add it to this program's skins options. The second screen for atractmode is a low priority though, the important part is just having very fast song list browsing with only two arrows for page up and page down.

chundermike, 2022-11-11:
I'm used to creating skins now so for me they aren't technically difficult, just time consuming if you're tweaking a lot. So it's difficult to say if they are easy for someone starting out...I would hope not. Best thing to do is have a read of the user guide and then look at the existing skins.  But I would definitely recommend talking to Jay (who's posted here)...I'm sure he won't mind; he's a really good guy. He has created some fabulous skins and I would say he has a great deal of knowledge about it now.  But if course I'd still be more than happy to help if you have any questions once you get stuck in.

Cheers
Mike

BobbyBoyGaming, 2022-11-11:
Awesome thanks!!

jaythom1, 2022-11-11:
Sure, hit me up. I love making skins. I'm not sure on implementing the second monitor thing though so I'd have to defer that back to Mike, but I know what your talking about.

BobbyBoyGaming, 2022-11-15:
@jaythom1 
I now have a layout of how I will want stuff to get displayed and here is a drawing of of my layout, just a sketch/draft/plan. I wanted to ask if you use discord or if this is a good method to stay in contact with you regarding this topic. Thanks

@chundermike I was finally able to set up the HiFi-Berry DAC+XLR with RaspiOS and it did work with Fruitbox once already.
However, I shut the RPi off and then on again and I started getting an error.... again... 

The error on the terminal is:
"XDG_RUNTIME_DIR not set in the environment"
and within the fruitbox program, what happens the songs stop loading and then it says:
"ERROR: Couldn't reserve audio samples", and the only option is to close the program.

It did work once already. Do you have any idea what this error means, or can you point me in the right direction? I also asked at the Hi-Fi Berry forum, however the soundcard is working fine now, I can play mp3 files within RaspiOS via VLC media player without any issues, this however happens when I load Fruitbox.

BobbyBoyGaming, 2022-11-15:
@jaythom1 Oh ok, Oh ok, maybe a page number or a page count be helpful if the scroll-bar "alphabet location" is too hard. 
Regarding the code generation, I was thinking of along the lines of "autogenerated" codes that are page-dependent. ISRC sounds like asking for trouble if it is some type of data base that I have to scrub the codes from. I like the way the current skins autogenerate the 2-digit alphanumeric code and that the codes are page dependent such that "A7" on page 3 is one song and "A7 on page 4 is another song.

Regarding the fast page flipping, and current "title strips" skins. I feel like these "title-strip" matrices are very difficult to read. I kind of just get lost on the screen and it is not really clear to me why they always come in pairs, it is too easy for me to enter the wrong code and get the wrong song with this "title-strip" method. 

So this method might be closer to the reality of the classic jukeboxes but im not so interested in simulating an old jukebox, i am more interested in readability for first-time users. 

So with title-strip matrices, for example if I want a song by "The Who", it is very hard for my eyes to find "The Who" text on the screen, and then I have to figure out If I am reading the song from the top half of the strip or the bottom half of the strip. Then if the strip does not contain the song of interest, but I know for a fact there should be another strip for the same band, I have to figure out whether to jump to the next strip towards the right, or downward? Is the list written horizontally? is it written vertically? 

It is simply too much brain power is required to just get to the good stuff. So I want to avoid the title-strip style altogether. I hope that explanation makes sense. It is like reading a newspaper column? or like reading a book? It is just too confusing and too cluttered for my liking even if the responsiveness were increased by eliminating those timers.

One timer i do definitely want to eliminate is "auto page flipping" I want the thing to never change page by itself. I was using the jukebox briefly to get acquainted with what changes I want to make to it. and I noticed that if I take a bit long to enter a code, the page simply flips by itself?  Maybe I had a bug or something? IDK

Thanks for the discord, I joined it! if things get a little bit too detailed/off-topic/diverted I can switch to that or have it as backup thanks.

jaythom1, 2022-11-15:
OK, well you can definitely have a list style page, and the auto page turning (TitleStripAutoTurnTime) can be set to zero which basically turns it off. All the attract mode stuff can be turned off so the pages never turn on their own. That's not a problem at all.

The songs on the title strips don't always come in pairs. Take a look at the Woody's Jukebox skin. It only displays one artist and one song title per title strip. It even has different color for the artist, vs the song title. All of this stuff is easy to program into the skin.

Now @chundermike can correct me on any of this if I'm wrong, but I don't believe there is a function to generate page numbers or letters like you're asking for. Concerning the Code/ID, I didn't understand what you were asking for initially but I think what you want is have the number selections like on any other skin like A7. You press the buttons A7 and it plays the corresponding song. You want it to also be able to play A7 on page 8 for instance. You would want to press something like A78 and it plays A7 from page 8. I don't believe that is possible with fruitbox. You have to be looking at the page and click the buttons A7. It won't let you select a song from a page that is not visible on the screen. The software just doesn't work like that. Part of the issue when you said "im not so interested in simulating an old jukebox". That is what Fruitbox is designed to do, so straying from that can make things challenging.

I saw you say you wanted purely numeric codes. That is not an issue.

As far as page turning and finding the pages you want, you can set a key to turn the page to go to the next page in the alphabet. I simply look at the first title strip and see what letter the song starts with to figure out what page I'm on. So on my jukebox I built, I have page left, page right and _alpha_ left, _alpha_ right buttons. 

As for reading the title strips in existing skins, yes. In each column, they are sorted vertically. For me the hardest part of finding The Who is, does it start with a T or a W? 

Finding part 2 of a song should be as easy as selecting the song right after part 1, assuming your songs are named correctly or the ID3tags are correct. 

I would still need to know what size monitor you are planning to use and what the resolution is. This is important for me to figure out what size text to use so we can cram as many songs per page and still be readable.

Just curious, but have you got Fruitbox running on a Pi3 yet, or are you still just planning?

chundermike, 2022-11-15:
Thanks Jay for answering the questions so well.  Just a few things to add...

The contents of title strips is completely flexible... the *Description parameters in the [general] section of the skin cfg file allow any text and/or field to be specified for the artist and song title, as well as adding extra iines carrying whatever other information you may want.  And because title strips can be placed anywhere on the screen, they can be arranged horizontally if needed.  Also, yes, you don’t need to have song title paired on each title strip – there’s a PairSongs parameter in the skin cfg file which determines that.

You can’t display letters for the current visible pages (as they can be sorted lots of ways not just alphabetically), but you can display the page number using status variables such as {(FIRST_VISIBLE_TITLE_STRIP} for example.  A good skin example is “Terminal”, which has quite a few status fields on display.

There’s an ini file parameter in the [general] section called IgnoreArtistPrefixThe, which ignores the beginning ‘The’ from artist names when sorting, so for example “The Who” is placed correctly in the “W” section of the song list.

Finally, don’t forget, there’s always the search feature (accessed by pressing forward slash on the keyboard), which allows you to filter out and show songs with particular titles, artists, years, genre, etc.

Cheers,
Mike

BobbyBoyGaming, 2022-11-16:
@jaythom1 sorry for the confusion about the song codes part. Yes, what I meant is that this (which you said)
"You have to be looking at the page and click the buttons A7." is totally fine. Just have a fixed number of lines of text on each screen, and then however many numbers fit in there from 01 to 99 for example, and then as I navigate the jukebox interface I am just reusing these codese which are page/context dependent. 
So, basically how it already works is fine, but instead of a letter and a number, just two numbers. 

I dont know the size of the monitor that I will be using but it is 1080p, 16:9 aspect ratio as far as I know. I dont have the monitor with me at the moment, but I am fairly confident it is a 17'' monitor measured diagonally.
[image]
Here is a picture of the model cabinet I have, but I basically gutted the whole thing and its just a cabinet. so I put a monitor on the inside where the coverart-flipping mechanism was. 

This is a legit jukebox, but it seems like fruitbox is only simulating jukeboxes from the 50s, this is like more "modern", from the 80s since it was a CD-R jukebox.

The jukebox is currently running with a different system shown here: https://www.cdadapter.com/nsm4mp3.htm 
However, this system has limitation as it only replaces the CD-changer but it still operates off-of the stock motherboard. I had to separately create a song-list display using a small LCD screen with its own "intel-compute-module"-knock off that basically boots into a PDF list of the songs. However this list needs to be created in a very tedious and specific way so that it is compatible with the current system.

With my new Upgrade based off of fruitbox + HiFi Berry, I will be able to get better sound quality, and more configurability. If I can get it all to work perfectly of course.

Regarding your last question, I already have a RPi4 with Fruitbox V2 running, it has been tested with HiFi-Berry DAC+XLR and a Pyle 1000Watt Amp and everything seems to be working (with exception of an Error I have to ask Mike about). It is not inside the cabinet though, I am just running it on my desktop at home as I learn more, test things, create a new skin, load music, and all of that. I did have it on an RPi 3B+ but I ditched it now. 

- - - - - -

@chundermike Great thanks I took note of all those commands I will be playing around with them for sure. 

However, I do have one pressing issue at the moment:
[image]

The error stopped appearing once and then it returned, and stayed. I have no idea what to do or what this error means. Do you have any idea? Thanks!!!