chundermike/rpi-fruitbox-v2 #21: Raspian 10 Buster 32bit  - Missing GLIBC_2.29

batfink5000, 2022-11-16:
Hi.

Is V2 of fruitbox supported on Buster ?
I have installed the 32 bit version but get the following error when trying to run.

./fruitbox: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.29' not found (required by ./fruitbox)

Thanks
BF.

For info I came across the following items during the install which I was able to work around.
* The installer fails unless executed in the home dir ( /home/pi ).

* My system was missing /usr/lib/arm-linux-gnueabihf/libbcm_host.so.0 but I was able to resolve this :
  sudo ln -s /opt/vc/lib/libbcm_host.so /usr/lib/arm-linux-gnueabihf/libbcm_host.so.0

* My system was missing libOpenGL.so.0 but able to resolve this:
   sudo apt install libopengl0 -y

chundermike, 2022-11-16:
Hi

I built it on Bullseye, looks like Buster is different... :(

chundermike, 2022-11-16:
Also only works on the Desktop version and not Lite.

batfink5000, 2022-11-16:
I believe I have the  Desktop version of Buster.
Would it make sense for me to wait and try to compile it against buster when you release the source ?

batfink5000, 2022-11-16:
Great, thanks for your support.  Much appreciated.

chundermike, 2022-11-17:
Source code now available - see build directory, and build.txt

batfink5000, 2022-11-17:
Hi.
Thanks for that.
I followed the build.txt which worked perfectly except for one detail.
In the txt, it refers to the v1 repo (wget https://codeload.github.com/chundermike/rpi-fruitbox/zip/master), rather than v2.
Apart from that no issues.

Thanks again.
BF

chundermike, 2022-11-17:
Ah thanks for spotting that...I'll correct it later. Glad to know it worked for you :)

Cheers
Mike

closed 2022-11-18 by chundermike

CaMer0n, 2023-01-14:
@batfink5000 I'm using Buster on a 32bit Pi4 and have the same issue.  I tried the build instructions, but it ended with an error: 
```
/usr/bin/ld: spectrum_analyser.o: in function `SpectrumAnalyserDisplayClass::update(unsigned int)':
spectrum_analyser.cpp:(.text+0x510): undefined reference to `spectrum_analyser_band'
collect2: error: ld returned 1 exit status
make: *** [Makefile_pi4_32.mak:17: ../rpi4_32/fruitbox] Error 1
```
Any thoughts? 
btw, are you familiar with this? I wonder if it would be easy to adapt for v2?
https://github.com/tamusiunas/retropie-fruitbox/blob/main/install-fruitbox-on-retropie.bash

chundermike, 2023-01-14:
Hi

Did you copy the equaliser.c into the right place? Check the build.txt instruction file carefully.

Cheers
Mike

https://github.com/tamusiunas/retropie-fruitbox/blob/main/install-fruitbox-on-retropie.bash

CaMer0n, 2023-01-15:
Thank you!!. I ran the build again and that error is gone. Now I just need to get past the "Failed to get display mode". 
I'm running it from the Retropie menu.