chundermike/rpi-fruitbox-v2 #22: Installing from binaries

vigorator, 2022-11-19:
Hi,
Im trying to install fruitbox from binaries on my Ubuntu.
After getting and fixing some errors with missing files, I was able to run a make but I got an error:

 make -f Makefile_pi4.mak
g++ -c -o fruitbox.o fruitbox.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
g++ -c -o display_base.o display_base.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
g++ -c -o playqueue.o playqueue.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
g++ -c -o audio.o audio.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
audio.cpp: In static member function ‘static off_t AudioClass::file_lseek_cb(void*, off_t, int)’:
audio.cpp:630:45: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘off_t’ {aka ‘long int’} [-Wformat=]
  630 |   if(ret < 0) fprintf(stderr, "seek error (%d, %d): %s\n", ret, errno, strerror(errno));
      |                                            ~^              ~~~
      |                                             |              |
      |                                             int            off_t {aka long int}
      |                                            %ld
audio.cpp: In member function ‘void AudioClass::PlaySound(sound_t&)’:
audio.cpp:1008:55: warning: narrowing conversion of ‘((double)Config->ConfigClass::ini_general->IniGeneralConfigClass::song_volume / 1.0e+2)’ from ‘double’ to ‘float’ [-Wnarrowing]
 1008 |   float master_vol { Config->ini_general->song_volume / 100.0 };
      |                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
g++ -c -o config.o config.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
g++ -c -o database.o database.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
g++ -c -o display.o display.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
g++ -c -o engine.o engine.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
engine.cpp: In member function ‘void EngineClass::SetButtonBit(uint32_t)’:
engine.cpp:460:23: warning: narrowing conversion of ‘(b & 1)’ from ‘uint64_t’ {aka ‘long unsigned int’} to ‘uint32_t’ {aka ‘unsigned int’} [-Wnarrowing]
  460 | #define BPM_IDX(A) (A & static_cast<uint64_t>(1))
      |                    ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
engine.cpp:468:18: note: in expansion of macro ‘BPM_IDX’
  468 |   uint32_t idx { BPM_IDX(b) };
      |                  ^~~~~~~
g++ -c -o title_strips.o title_strips.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
g++ -c -o status.o status.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
g++ -c -o userguide.o userguide.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
g++ -c -o bitmap.o bitmap.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
g++ -c -o joystick.o joystick.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
g++ -c -o mouse.o mouse.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
g++ -c -o input.o input.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
input.cpp: In member function ‘void InputClass::scanDevice(input_device_t&, input_event_t&)’:
input.cpp:409:15: warning: narrowing conversion of ‘read(device.input_device_t::fd, ((void*)((input_event*)(& ev))), (sizeof (input_event) * 64))’ from ‘ssize_t’ {aka ‘long int’} to ‘int’ [-Wnarrowing]
  409 |   int rb {read(device.fd, ev, sizeof(struct input_event)*64)};
      |           ~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
input.cpp:702:58: warning: narrowing conversion of ‘Config->ConfigClass::skin_touch_song->TouchSongDisplayClass::<anonymous>.JoystickDisplayClass::<anonymous>.DisplayBaseClass::size.dynamics2_t::x.dynamic_t::val’ from ‘int32_t’ {aka ‘int’} to ‘uint32_t’ {aka ‘unsigned int’} [-Wnarrowing]
  702 |             uint32_t sw {Config->skin_touch_song->size.x.val};
      |                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
input.cpp:703:58: warning: narrowing conversion of ‘Config->ConfigClass::skin_touch_song->TouchSongDisplayClass::<anonymous>.JoystickDisplayClass::<anonymous>.DisplayBaseClass::size.dynamics2_t::y.dynamic_t::val’ from ‘int32_t’ {aka ‘int’} to ‘uint32_t’ {aka ‘unsigned int’} [-Wnarrowing]
  703 |             uint32_t sh {Config->skin_touch_song->size.y.val};
      |                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
g++ -c -o gpio.o gpio.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
g++ -c -o dialog.o dialog.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
dialog.cpp: In member function ‘bool DialogCalibrateTouchClass::Run()’:
dialog.cpp:1692:19: warning: narrowing conversion of ‘(btn_press & 16383)’ from ‘uint32_t’ {aka ‘unsigned int’} to ‘int32_t’ {aka ‘int’} [-Wnarrowing]
 1692 |         btn_press & BUTTON_RAW_XTOUCH_MASK,
dialog.cpp:1693:48: warning: narrowing conversion of ‘((btn_press >> 14) & 16383)’ from ‘uint32_t’ {aka ‘unsigned int’} to ‘int32_t’ {aka ‘int’} [-Wnarrowing]
 1693 |         (btn_press >> BUTTON_RAW_YTOUCH_SHIFT) & BUTTON_RAW_YTOUCH_MASK
dialog.cpp: In member function ‘bool DialogSoftwareUpdateClass::Run()’:
dialog.cpp:1991:15: warning: ignoring return value of ‘int system(const char*)’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
 1991 |         system(system_cmd.c_str());
      |         ~~~~~~^~~~~~~~~~~~~~~~~~~~
dialog.cpp:2024:17: warning: ignoring return value of ‘int system(const char*)’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
 2024 |           system(system_cmd.c_str());
      |           ~~~~~~^~~~~~~~~~~~~~~~~~~~
dialog.cpp:2036:17: warning: ignoring return value of ‘int system(const char*)’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
 2036 |           system(system_cmd.c_str());
      |           ~~~~~~^~~~~~~~~~~~~~~~~~~~
dialog.cpp:2041:17: warning: ignoring return value of ‘int system(const char*)’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
 2041 |           system(system_cmd.c_str());
      |           ~~~~~~^~~~~~~~~~~~~~~~~~~~
g++ -c -o spectrum_analyser.o spectrum_analyser.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
spectrum_analyser.cpp: In member function ‘virtual bool SpectrumAnalyserDisplayClass::update(uint32_t)’:
spectrum_analyser.cpp:59:33: warning: narrowing conversion of ‘((SpectrumAnalyserDisplayClass*)this)->SpectrumAnalyserDisplayClass::band.std::vector<band_gain_t>::size()’ from ‘std::vector<band_gain_t>::size_type’ {aka ‘long unsigned int’} to ‘uint32_t’ {aka ‘unsigned int’} [-Wnarrowing]
   59 |   uint32_t num_bands { band.size() };
      |                        ~~~~~~~~~^~
g++ -c -o fast_random.o fast_random.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
g++ -c -o menus.o menus.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
g++ -c -o touch_areas.o touch_areas.cpp -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -I/opt/vc/include
g++ -Wall -o ../rpi4/fruitbox fruitbox.o display_base.o playqueue.o audio.o config.o database.o display.o engine.o title_strips.o status.o userguide.o bitmap.o joystick.o mouse.o input.o gpio.o dialog.o spectrum_analyser.o fast_random.o menus.o touch_areas.o -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -lmpg123 -lm -ldl -L/opt/vc/lib/ -lcurl -lbcm_host -lpthread `pkg-config --libs --static allegro-static-5 allegro_memfile-static-5 allegro_font-static-5 allegro_ttf-static-5 allegro_primitives-static-5 allegro_image-static-5 allegro_audio-static-5 allegro_acodec-static-5 allegro_video-static-5`
/usr/bin/ld: skipping incompatible /opt/vc/lib//libbcm_host.so when searching for -lbcm_host
/usr/bin/ld: cannot find -lbcm_host: No such file or directory
/usr/bin/ld: skipping incompatible /opt/vc/lib//libbcm_host.so when searching for -lbcm_host
collect2: error: ld returned 1 exit status
make: *** [Makefile_pi4.mak:17: ../rpi4/fruitbox] Error 1

Any ideas what i missed?

chundermike, 2022-11-19:
Hi,

Looks like the compilation was successful, but the link failed.

It's trying to link in the bcm_host library, but this is a Raspberry-Pi specific library, so won’t be found.  This is my bad, as I should have said you need to remove this for non-rpi target hardware.  The _RPI directive only affects the source code, not the make file.

So, in the make file, remove any mention of bcm_host (I.e. remove -lbcm_host).

Hopefully that should do the trick...

Cheers,
Mike

input.cpp:702:58: warning: narrowing conversion of ‘Config->ConfigClass::skin_touch_song->TouchSongDisplayClass::.JoystickDisplayClass::.DisplayBaseClass::size.dynamics2_t::x.dynamic_t::val’ from ‘int32_t’ {aka ‘int’} to ‘uint32_t’ {aka ‘unsigned int’} [-Wnarrowing]
input.cpp:703:58: warning: narrowing conversion of ‘Config->ConfigClass::skin_touch_song->TouchSongDisplayClass::.JoystickDisplayClass::.DisplayBaseClass::size.dynamics2_t::y.dynamic_t::val’ from ‘int32_t’ {aka ‘int’} to ‘uint32_t’ {aka ‘unsigned int’} [-Wnarrowing]

chundermike, 2022-11-20:
The compilation was successful, but the link failed, due to it not being able to find the bcm_host library.  The bcm_host library is Pi hardware specific, sorry I forgot to mention this. 

Remove the -lbcm_host from the make file and try again.

Mike

vigorator, 2022-11-21:
Sorry, I should say on the beginning that Ive never compiled anything so in this case I'm very beginner :/

~/fruitbox/build$ make -f Makefile_pi4.mak
g++ -Wall -o ../rpi4/fruitbox fruitbox.o display_base.o playqueue.o audio.o config.o database.o display.o engine.o title_strips.o status.o userguide.o bitmap.o joystick.o mouse.o input.o gpio.o dialog.o spectrum_analyser.o fast_random.o menus.o touch_areas.o -O3 -g0 -std=c++17 -D_RPI -D_RPI4 -lmpg123 -lm -ldl -L/opt/vc/lib/ -lcurl -lpthread `pkg-config --libs --static allegro-static-5 allegro_memfile-static-5 allegro_font-static-5 allegro_ttf-static-5 allegro_primitives-static-5 allegro_image-static-5 allegro_audio-static-5 allegro_acodec-static-5 allegro_video-static-5`
/usr/bin/ld: spectrum_analyser.o: warning: relocation against `spectrum_analyser_band' in read-only section `.text'
/usr/bin/ld: display_base.o: in function `DynamicParameter::{lambda(void*)#9}::_FUN(DynamicParameter)':
display_base.cpp:(.text._ZN16DynamicParameterMUlPvE7_4_FUNES_[_ZN16DynamicParameterMUlPvE7_4_FUNES_]+0x19): undefined reference to `spectrum_analyser_band'
/usr/bin/ld: display_base.o: in function `DynamicParameter::{lambda(void*)#10}::_FUN(DynamicParameter)':
display_base.cpp:(.text._ZN16DynamicParameterMUlPvE8_4_FUNES_[_ZN16DynamicParameterMUlPvE8_4_FUNES_]+0x19): undefined reference to `spectrum_analyser_band'
/usr/bin/ld: display_base.o: in function `DynamicParameter::{lambda(void*)#11}::_FUN(DynamicParameter)':
display_base.cpp:(.text._ZN16DynamicParameterMUlPvE9_4_FUNES_[_ZN16DynamicParameterMUlPvE9_4_FUNES_]+0x19): undefined reference to `spectrum_analyser_band'
/usr/bin/ld: audio.o: in function `AudioClass::Run()':
audio.cpp:(.text+0x1057): undefined reference to `init_equalizer'
/usr/bin/ld: audio.o: in function `AudioClass::PlaySongThread()':
audio.cpp:(.text+0x255c): undefined reference to `clear_equalizer'
/usr/bin/ld: gpio.o: in function `GpioClass::OpenDevice()':
gpio.cpp:(.text+0xee): undefined reference to `bcm_host_get_peripheral_address'
/usr/bin/ld: gpio.o: in function `GpioClass::ConfigurePinAsInput(unsigned int)':
gpio.cpp:(.text+0x3b7): undefined reference to `bcm_host_get_peripheral_address'
/usr/bin/ld: gpio.o: in function `GpioClass::ConfigurePinAsOutput(unsigned int)':
gpio.cpp:(.text+0x597): undefined reference to `bcm_host_get_peripheral_address'
/usr/bin/ld: spectrum_analyser.o: in function `SpectrumAnalyserDisplayClass::update(unsigned int)':
spectrum_analyser.cpp:(.text+0x13c): undefined reference to `spectrum_analyser_band'
/usr/bin/ld: spectrum_analyser.cpp:(.text+0x286): undefined reference to `spectrum_analyser_band'
/usr/bin/ld: spectrum_analyser.cpp:(.text+0x51b): undefined reference to `spectrum_analyser_band'
/usr/bin/ld: spectrum_analyser.cpp:(.text+0x58b): undefined reference to `spectrum_analyser_band'
/usr/bin/ld: warning: creating DT_TEXTREL in a PIE
collect2: error: ld returned 1 exit status
make: *** [Makefile_pi4.mak:17: ../rpi4/fruitbox] Error 1

chundermike, 2022-11-21:
And you need to remove the -D_RPI as well as the -lbcm_host from the make file

vigorator, 2022-11-21:
I did everything from scratch, make file updated, equaliser.c copied.
still issue with undefined reference to 'spectrum_analyser_band'

`make -f Makefile_pi4.mak
g++ -c -o fruitbox.o fruitbox.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
g++ -c -o display_base.o display_base.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
g++ -c -o playqueue.o playqueue.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
g++ -c -o audio.o audio.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
audio.cpp: In static member function ‘static off_t AudioClass::file_lseek_cb(void*, off_t, int)’:
audio.cpp:630:45: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘off_t’ {aka ‘long int’} [-Wformat=]
  630 |   if(ret < 0) fprintf(stderr, "seek error (%d, %d): %s\n", ret, errno, strerror(errno));
      |                                            ~^              ~~~
      |                                             |              |
      |                                             int            off_t {aka long int}
      |                                            %ld
audio.cpp: In member function ‘void AudioClass::PlaySound(sound_t&)’:
audio.cpp:1008:55: warning: narrowing conversion of ‘((double)Config->ConfigClass::ini_general->IniGeneralConfigClass::song_volume / 1.0e+2)’ from ‘double’ to ‘float’ [-Wnarrowing]
 1008 |   float master_vol { Config->ini_general->song_volume / 100.0 };
      |                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
g++ -c -o config.o config.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
g++ -c -o database.o database.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
g++ -c -o display.o display.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
g++ -c -o engine.o engine.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
engine.cpp: In member function ‘void EngineClass::SetButtonBit(uint32_t)’:
engine.cpp:460:23: warning: narrowing conversion of ‘(b & 1)’ from ‘uint64_t’ {aka ‘long unsigned int’} to ‘uint32_t’ {aka ‘unsigned int’} [-Wnarrowing]
  460 | #define BPM_IDX(A) (A & static_cast<uint64_t>(1))
      |                    ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
engine.cpp:468:18: note: in expansion of macro ‘BPM_IDX’
  468 |   uint32_t idx { BPM_IDX(b) };
      |                  ^~~~~~~
g++ -c -o title_strips.o title_strips.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
g++ -c -o status.o status.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
g++ -c -o userguide.o userguide.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
g++ -c -o bitmap.o bitmap.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
g++ -c -o joystick.o joystick.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
g++ -c -o mouse.o mouse.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
g++ -c -o input.o input.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
input.cpp: In member function ‘void InputClass::scanDevice(input_device_t&, input_event_t&)’:
input.cpp:409:15: warning: narrowing conversion of ‘read(device.input_device_t::fd, ((void*)((input_event*)(& ev))), (sizeof (input_event) * 64))’ from ‘ssize_t’ {aka ‘long int’} to ‘int’ [-Wnarrowing]
  409 |   int rb {read(device.fd, ev, sizeof(struct input_event)*64)};
      |           ~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
input.cpp:702:58: warning: narrowing conversion of ‘Config->ConfigClass::skin_touch_song->TouchSongDisplayClass::<anonymous>.JoystickDisplayClass::<anonymous>.DisplayBaseClass::size.dynamics2_t::x.dynamic_t::val’ from ‘int32_t’ {aka ‘int’} to ‘uint32_t’ {aka ‘unsigned int’} [-Wnarrowing]
  702 |             uint32_t sw {Config->skin_touch_song->size.x.val};
      |                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
input.cpp:703:58: warning: narrowing conversion of ‘Config->ConfigClass::skin_touch_song->TouchSongDisplayClass::<anonymous>.JoystickDisplayClass::<anonymous>.DisplayBaseClass::size.dynamics2_t::y.dynamic_t::val’ from ‘int32_t’ {aka ‘int’} to ‘uint32_t’ {aka ‘unsigned int’} [-Wnarrowing]
  703 |             uint32_t sh {Config->skin_touch_song->size.y.val};
      |                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
g++ -c -o gpio.o gpio.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
g++ -c -o dialog.o dialog.cpp -O3 -g0 -std=c++17 -D_RPI4 -I/opt/vc/include
dialog.cpp: In member function ‘bool DialogCalibrateTouchClass::Run()’:
dialog.cpp:1692:19: warning: narrowing conversion of ‘(btn_press & 16383)’ from ‘uint32_t’ {aka ‘unsigned int’} to ‘int32_t’ {aka ‘int’} [-Wnarrowing]
 1692 |         btn_press & BUTTON_RAW_XTOUCH_MASK,
dialog.cpp:1693:48: warning: narrowing conversion of ‘((btn_press >> 14) & 16383)’ from ‘uint32_t’ {aka ‘unsigned int’} to ‘int32_t’ {aka ‘int’} [-Wnarrowing]
 1693 |         (btn_press >> BUTTON_RAW_YTOUCH_SHIFT) & BUTTON_RAW_YTOUCH_MASK
dialog.cpp: In member function ‘bool DialogSoftwareUpdateClass::Run()’:
dialog.cpp:1991:15: warning: ignoring return value of ‘int system(const char*)’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
 1991 |         system(system_cmd.c_str());
      |         ~~~~~~^~~~~~~~~~~~~~~~~~~~
dialog.cpp:2024:17: warning: ignoring return value of ‘int system(const char*)’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
 2024 |           system(system_cmd.c_str());
      |           ~~~~~~^~~~~~~~~~~~~~~~~~~~
dialog.cpp:2036:17: warning: ignoring return value of ‘int system(const char*)’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
 2036 |           system(system_cmd.c_str());
[... cut ...]
spectrum_analyser.cpp:(.text+0x13c): undefined reference to `spectrum_analyser_band'
/usr/bin/ld: spectrum_analyser.cpp:(.text+0x286): undefined reference to `spectrum_analyser_band'
/usr/bin/ld: spectrum_analyser.cpp:(.text+0x51b): undefined reference to `spectrum_analyser_band'
/usr/bin/ld: spectrum_analyser.cpp:(.text+0x58b): undefined reference to `spectrum_analyser_band'
/usr/bin/ld: warning: creating DT_TEXTREL in a PIE
collect2: error: ld returned 1 exit status
make: *** [Makefile_pi4.mak:17: ../rpi4/fruitbox] Error 1`

chundermike, 2022-11-21:
Assuming after you copied equaliser.c you then recompiled and installed the mpg123 library and then recompiled fruitbox?

vigorator, 2022-11-21:
I did it from scratch like in built.txt:
tar libmpg123
cp equalizer.c
make & sudo make of libmpg123

then built and make liballegro

edit file Makefile_pi4.mak (remove -D_RPI and  -lbcm_host)
built fruitbox with these errors :/

chundermike, 2022-11-21:
Ok, I will have a closer look this evening :)

vigorator, 2022-11-21:
I can give you access to my machine. if will be easier for you.

chundermike, 2022-11-21:
Hi, I've only got a Ubuntu Virtual Machine on my Windows PC, but managed to get it to build.  There are however problems - it doesn't seem to recognise the button presses, and doesn't run full-screen.  Not sure if this is an issue with my VM, or Ubuntu.  Anyway, try a different cpu target when compiling the mpg123 library...

```
./configure --with-cpu=generic_fpu --disable-shared
make && sudo make install

```
...then rebuild fruitbox.

Let me know if this works.  If it doesn't I'll send you my ubuntu fruitbox executable.  If it does work, let me know if the button input is broken, like it is on mine.  If so, I will have to spend some time debugging.

vigorator, 2022-11-21:
Got some errors regarding gpios even if I removed all gpio references from make file

My email is kkorpys@gmailc.om

chundermike, 2022-11-21:
Ok, I've emailed you the executable :)

feivel2211, 2023-05-23:
Hi, im having the same issue here. Like @vigorator i tried to compile on a Debian on a physical Notebook, i got the same error complaining about spectrum analyzer.

So i tried the way like @chundermike did, compiling on a Ubuntu VM. That worked.  

Right now i have the same issue as you @chundermike. Keyboard / Mouse input is blocked when starting Fruitbox. Thats happening on the VMWare VM, aswell on the physical Notebook i installed with Ubuntu.

chundermike, 2023-05-23:
Hi

If input is blocked, it may be that you don’t have permissions on the input devices. From a command line, try

sudo chmod 777 /proc/bus/input/devices

Mike

feivel2211, 2023-05-23:
Hello mike, thank you for you fast reply. 
i tried _

sudo chmod 777 /proc/bus/input/devices_ 

but it did not make any difference

Oliver

chundermike, 2023-05-23:
Hi

Sorry it may be a different command like

sudo chmod 777 /dev/input/event*

Mike

feivel2211, 2023-05-23:
Thank you Mike! 

sudo chmod 777 /dev/input/event* solved the blocked input for me!

Oliver

chundermike, 2023-05-23:
Great!

Do you have any other issues?

Mike

feivel2211, 2023-05-23:
Maybe you can give me a hint how to get it in Kiosk Mode.. Been looking through all flags and ini setting but the taskbar and the op bar is always in foreground... 
[image]

chundermike, 2023-05-23:
I think that’s a setting under Ubuntu, nothing to do with fruitbox. Try your desktop settings…it may be something like auto-hide ?

Mike

[2023-05-23 13_43_14-Foto - Google Fotos]<https://user-images.githubusercontent.com/134369379/240261298-224f2568-789f-423d-bec6-c2439d2e6b42.png>

feivel2211, 2023-05-23:
You´re right. I´ve set the Dock on the left side to auto hide, for hiding the top bar i installed an ubuntu plugin. Thank you!