chundermike/rpi-fruitbox-v2 #23: USB party mode

beit-no1, 2022-11-20:
Hi Mike,
Not sure this is an issue, but I'm trying to see how I can setup my JukeBox in a party mode.  I want to have it configured as now with a base set of songs and music, but I'd like to have various other USB keys setup with Genre of songs or allow people to bring their own.  I have two USB ports at the front connected to the Raspberry Pi 3B+ and want someone to plug in the USB and then press a button wired into the GPIO which will essentially restart Fruitbox with the same skin, but use the USB and build the DB of songs on the USB key.  I'd be ok if it had to rebuild the key DB each time so I guess it doesn't have to be stored on the USB, but would need to be rebuilt each time as the user may have changed the songs.

Is this the right place to load this as a request as to whether you think something like this can be done with the current version.  Maybe it is more automation and scripting in the Linux environment, but thought I'd ask here before I start trying to fully set it up.   

The latest build has corrected the memory leak for me and so it's been stable for about 4 days running and I'm ready to now extend it.  I envision a set of keys next to the jukebox that people can switch so that it isn't one Library with too many pages..  If there is a different equally efficient way then I'd be open as I also have a large music library on my network, but felt then I'd need to get into sync'ng the music more local in genre libraries...  the USB key allows people to come prepared.

Thanks for your time to read.

chundermike, 2022-11-20:
Hi,

I would say that's a linux script issue...and that's something I'm not an expert on. But I would imagine it working something like this...

Instead of starting fruitbox directly, call it from a script. And the script would do this...
1. Check if a usb stick is plugged in (how I don't know...maybe try writing or reading a dummy file to the usb root to see if it's there? Or perhaps there's another command just to see if it's mounted?)
2. If there is a usb drive plugged in, then run fruitbox with the --music-path <usb root> command line option
3. But if there's no usb stick mounted, just run fruitbox as normal (and it'll pick up the music path(s) from the ini file instead, and load your music).

What do you think?

Mike

beit-no1, 2022-11-20:
Ok.. I already have a button setup for the restart on the RPi, so agreed that it's something at the startup.  I agree that your steps make sense and I'll check maybe that there are some music files on the plugged USB, but if yes then your approach is good.

I love the software, but have always thought having a filter on a switch/button would be great.  In my house my daughter, wife and I would all want to be able to switch quick to our JukeBox version...   rather than have all of our songs on one Jukebox... this is the high level ability to change the 100 songs I was thinking of.

Time to start programming in Linux it seems... I've done a lot of PowerShell so just going to have to figure out the startup commands.  This is great and feel free to close this issue as you've helped to send me in the right direction.

chundermike, 2022-11-20:
You're not the first I think to ask for real time switching between music sets. I don't think it'll be that difficult to implement, so maybe I'll add that to the to-do list. It'll be fun to code!

Maybe it'll work like this...

Currently you can specify more than one music path, so if I add a "tag" (say a descriptive text) to each music path setting, they can be grouped (i.e. "Dad's music", "Mum's music', etc.   Then have a dedicated button to cycle through each group.  Fruitbox will display the group name as you cycle through your choices.

Sound ok?

beit-no1, 2022-11-20:
This is exactly what I've been thinking and the USB is just my concept to tell people to bring their own party selection. 

Those 3 are my core and then can have synced some other folder groups for 80's or other generations. Nice to keep the jukebox to small number of pages. 

Will watch for this and would be happy to help test it while I start to code my usb solution with python.

chundermike, 2022-11-21:
Just had another couple of thoughts...

I will make it possible to have more than one group name associated with each music path, so that groups can share music.

Also, I think if you define a music path for the usb drive(s) and give it a unique group name, then you won't need a python script at all...just get fruitbox to select that music group using the new button, and all you'll see is the music on the USB drive and not your personal collections. The advantage of this is that if half-way through the night you realise how dreadful your guest's music taste is, you can quickly switch to your stuff without unplugging the usb drive and restarting fruitbox ;). Unfortunately their choices will still be queued up to play, but there's always the "Clear Queue" button to come to the rescue...

beit-no1, 2022-11-22:
Thanks.. not sure I fully understand though your thought(s).   Are you thinking that under the music path there would be different folders to designate the "groups" which are then translated into a drop-down or select which can be cycled with a button?  I can see that for the music on the system drive, but not sure how it helps with the USB.   If a user walks up and switches the USB then I am hoping to hit a button which will read the USB and ask to load it where it will create a new DB for that USB.  If more than one found it could ask which one.

That's how it could be within Fruitbox, but for me my idea was when the Raspberry pi is starting up I'd scan anything in the device ports considered external and if a USB with say Greater than 10 songs (mp3, etc..) are found it would select and launch fruitbox after deleting the database.... this would then load that USB as the current.   I guess if I found 2 USBs with more than 10 songs i'd have to choose or pop up some UI to let the user pick.

I'm just thinking of ways to make it very flexible..... friends have come by and really like it, but the default songs are not the ideal always :o).   I don't want thousands as paging is not good.

Thanks for listening and thinking....

chundermike, 2022-11-22:
So it'll be something like...
    MusicPath = /Music/RockAlbums Dad's music
    MusicPath = /Music/80sCompilations Mum's music, Party Time!
    MusicPath = /Music/OldStuff Mum's music, Dad's music
    MusicPath = /Music/NewStuff Kid's weird songs
    MusicPath = /Music/90sPartyAlbums Party Time!
    
    ...would create four music groups : "Dad's music", "Mum's music", "Kid's weird songs", and "Party Time!"
    
    Dad's music would comprise the songs from /Music/RockAlbums and /Music/OldStuff
    Mum's music would comprise the songs from /Music/80sCompilations and /Music/OldStuff
    etc.

If you create group(s) which poiint to the USB devices then they can be selected just like any other group

beit-no1, 2022-11-22:
ok..  I see now that you are letting them have common overlaps.  I guess the trick will be how to parse the groups from the paths as spaces in the group names might be interesting, but essentially it seems to work.

I guess if the USB devices are mounted then I can have that location in the music path as a group as you say.  The restart I have working, but I think that I'd need an override for the database as each time they switched it then I want for USB mount point to always rebuild the database as it might have changed content more likely than the other music paths.

I'm thinking that when you move to a group if one of them is a group flagged for rebuild then it does that to rebuild the USB which is located there.  Family would always have our music in more permanent storage attached, but the USB is for friends of family or special test song sets.  I think the time to let it rebuild will be worth the convenience as it's like loading a whole new set of albums or 45's into the jukebox :o).  

I started on my python script, but the tricky part I'm finding is detecting the USB drives and mounting them so that I can scan for songs and launch fruitbox....  I guess I could hard code some of it as I've found in Linux just tricky detecting and making mount points as I guess that there is FAT, FAT32, NFTS, etc..... I'll get there, but was trying to find some code snipets to help as i've setup Visual Studio code and can run and build python remotely.... very cool :o) (not a developer in real life, but I manage developers and technology processes)

beit-no1, 2022-11-22:
Agreed... means as the user configuring I can put in an escape for a space I want to keep.... got it and that part is like you said "controllable".  I certainly am liking the concept.

The reason I wanted a flag is that I suspect the users with the USB will change the songs a lot so just felt best to rebuild it rather than make them delete it as I could hear them complaining... so was thinking of a flag that says "always delete database file" for paths considered removable storage (USB).  I do like the idea that it's a db in each directory... hence why I just think on removable storage each time it is loaded then it should be rebuilt.  If I'm putting files on the permanent storage then I know to remove the db to let it rebuild.  

There is no detection for it's changed and the solution is to have the user remove the small db files in the removable storage correct....  It's why on my python I was going to remove that as I launched fruitbox to assure it gets rebuilt (my concept, but I agree that there are pros and cons... hence the flag / setting).

chundermike, 2022-11-22:
I'm not sure we need the flag...fruitbox will look at the timestamp of the db file(s), and if they are older than the timestamp(s) of the directory they're in (which get modified whenever the contents changes, i.e. when songs are added or removed), then it'll regenerate that local db file. So it should all happen automatically...

If someone brings a new USB stick along which fruitbox hasn't seen before, there will be no db file(s) on the usb stick, so fruitbox will create them. If that person then goes home, comes back another time having modified their songs, then fruitbox will detect the db files(s) on the usb stick are out of date, and regenerate them automatically.  Your family db files somewhere else won't be affected.

beit-no1, 2022-11-22:
ok.. I was not aware of that change (maybe new to v2) but that I agree sounds great... I'll do some testing of it with a USB and a manual mountpoint. 

The use case makes sense and I agree that this sounds workable and I just need to setup the two manual mount points for the USB ports and then with this grouping I can make a USB-Top group and a USB-Bottom group (as I have two ports).   

I presume that if I had this group included in mine and there was no USB the exception handling just skips a missing group where the path doesn't exist....   if you do code this and want some help testing then please reach out.

chundermike, 2022-11-22:
The auto database update has been around since v1. And, if a musicpath points to somewhere that doesn't exist, then it should.just ignore that path. fruitbox will only complain if it can't find any songs at all, so if you have your private songs included then that should be fine.

beit-no1, 2022-11-22:
ok.. swear that at one point I felt I had to add songs and delete it for rebuilt.. maybe my bad testing or got caught into thinking that.  

Can we have a flag to say no songs found is fine or maybe this is an optional thing for "removable" paths as this would work then for the USB which maybe exists but has no songs....  not sure what the "complaint" will be, but I guess it's more an issue if all referenced music paths had no music as then you have an empty UI :)  I'm thinking to have the USB a reference and then it either has stuff or not.... could be an extra group for my songs or a stand-alone when it's in what I'm thinking is "party mode" :)