chundermike/rpi-fruitbox-v2 #24: GPIO setup

sotchris, 2022-11-21:
I have fruitbox V2.04 installed and I can't set the GPIO pins. Can someone tell me what changes I need to make in the fruibox.ini file. All the choices I have made so far have not worked.

Thanks

chundermike, 2022-11-21:
Hi,

Is this for button input?  If so, just connect your button contacts between the GPIO and GND, then run fruitbox, enter the menu (TAB on the keyboard), choose configure buttons, use left/right to.find the button you want to configure from the list, press Enter, then press the GPIO button you want to map that input to.  Note not all GPIO inputs are supported, especially if you have any HAT boards fitted. When you exit fruitbox the ini file should be updated with your button mapping. You can edit that ini file manually later if you like.

sotchris, 2022-11-21:
I have done this, but the file fruitbox.ini it doesn't change.. I tried 2 pins BCM 23 and 24 for input but it doesn't respond.
I try to fix the fruitbox.ini manually. in the fruitbox.log file it appears that Opened GPIO.

Thanks

chundermike, 2022-11-21:
Ok, I'll double check the GPIO stuff this evening.  Could be a bug.

sotchris, 2022-11-21:
ok Thanks

sotchris, 2022-11-21:
when I run sudo ./fruitbox, it returns an error message. Error: XDG_RUNTIME_DR not set in the environment. ERROR: couldn't reserve audio samples. Is it normal?

chundermike, 2022-11-21:
No, that's not normal. Are you on pi4 desktop?

sotchris, 2022-11-21:
I use pi400

sotchris, 2022-11-21:
I changed the pins 23 and 24 to pin8 and pin 7 and the changes worked

chundermike, 2022-11-21:
Ah that's good. I'll check later which GPIO I exclude

sotchris, 2022-11-21:
ok thanks

sotchris, 2022-11-21:
I did some tests and only the pin 4,5,6,7,8,14,15 they work. This is very little pin about the function of the fruitbox.
If there is any other idea

Thanks

chundermike, 2022-11-21:
Hi, maybe there are some interfaces (SPI, etc.) enabled in your desktop config? I'm not sure.

You could always use a USB dragonrise zero delay controller...they're cheap and give you lots of connections.

chundermike, 2022-11-21:
Yes, just use the configure buttons menu.

This setup it the same way as GPIO?

sotchris, 2022-11-21:
my desktop is GNU 8.32
GNU GPLv3+

sotchris, 2022-11-21:
I appreciate you

Thanks

BobbyBoyGaming, 2022-11-22:
> when I run sudo ./fruitbox, it returns an error message. Error: XDG_RUNTIME_DR not set in the environment. ERROR: couldn't reserve audio samples. Is it normal?

I get that error too

BobbyBoyGaming, 2022-11-26:
I will do a fresh install of the fruitbox v2, latest version, but then completely block all RaspiOS updates, and see if thse updates have any negative effect in causing this error. I cannot think of anything else, but clearly it has todo with the GPIO because I am using the HiFiBerry DACPlusXLR which also uses the GPIO.
The fact that you @sotchris are getting the same error may be helpful to solve this issue since you are also trying to use the GPIO for a very different reason

sotchris, 2022-11-26:
I have upgraded both the Raspberry  and the fruitbox v2.04 again but it keeps giving me the same error.

sotchris, 2022-11-26:
I have also ordered a USB controller to bypass the GPIO. I'm still thinking of using an old keyboard for Buttons