chundermike/rpi-fruitbox-v2 #25: Running DC or stepper motor

bennyboyle, 2022-12-09:
Hey Mike,

Thanks for all the hard work! I have been planning on building a jukebox for a while, and have an idea of running a DC or stepper motor with an actual spinning record while a song is playing.

I understand that I can tell a GPIO pin to be either HIGH or LOW based on specific criteria, such as a song playing (status output).

I’m pretty new to this, so this might be an easy task, but anyway.. Is it possible to start and stop a motor using a motor driver, based on these commands from Fruitbox?

chundermike, 2023-01-07:
Hi,

I'm not 100% sure I'm afraid. Fruitbox, as you said can drive gpio lines high and low, and you can use timers and other status to do this under different conditions, but I wouldn't rely on the timing if you need to do it fast, for instance to generate specific length pulses for the motor. 

Mike

bennyboyle, 2024-05-06:
**Solution:**

I bought a 28BYJ-48 stepper motor and a ULN2003AN board that I'm going to use for this.
Stepper motor running at 9 rpm with a gear ratio of 40 (motor) / 8 (record), the 7" record will spin at 45 rpm.

I had to write two codes in Python for this to work. A person who actually knows what he's doing could probably make this into one script. However, Chat GPT helped me a lot by writing the basic code for me.

The scripts:
1. Script for simply running the stepper motor at a given speed (9 rpm) indefinitely.
2. Script that notices if a given GPIO pin (4 in my case) is high or low. Based on this, it will run script number one on high or stop it on low.

Next, I included GPIO hi and low parameters in the status object for when a song is playing (stops when unloading/loading new song).

Also made a separate status object for when I EXIT Fruitbox, which will stop the motor/script from running (set GPIO 4 to low), if it is in motion/song is playing.

Haven't made the physical gears or setup for the spinning record yet. Hoping it will look bad ass spinning behind the screen!