chundermike/rpi-fruitbox-v2 #27: Random track order

fadangalini, 2023-01-07:
Hi. Great software, thanks for making and sharing.

I have a very large song collection so want everything displayed in a random order.
In the skin configuration file (I'm using Wurly), I changed ‘SortSongsBy’ to random.
The songs are displayed in a random order, but always the same order.
Is there anyway to have the songs in a different random order after each boot?

Many Thanks

chundermike, 2023-01-07:
Hi,

Looks like I forgot to uniquely seed the randomisation at start up, which will give you different results each time. I'll add that in the next release.

Thanks for the feedback

Cheers
Mike

fadangalini, 2023-01-07:
That would be great. Thanks!

closed 2023-01-07 by chundermike