chundermike/rpi-fruitbox-v2 #29: Choose skin

Presto1970, 2023-01-20:
Hi Mike, on a new build I don’t have the option to choose a skin on boot - how do I make that happen? I’m comfortable with adding to the ini file once I know what I like.

chundermike, 2023-01-20:
In the ini file if there is more than one entry in the [skins] section then fruitbox will invoke the skin chooser automatically.

Presto1970, 2023-01-20:
Sorry no, not for me. If I put another entry with a different skin, I can only cycle between those two?

chundermike, 2023-01-20:
in the fruitbox.ini file, if you have something like...

[skins]
File = skins/SB-M100/fruitbox.cfg
File = skins/Wurly/fruitbox.cfg
File = skins/TouchOne/fruitbox.cfg
File = skins/Classic/fruitbox.cfg

...then the skin chooser will cycle through those four.

Presto1970, 2023-01-20:
Ok yes I get that but is there a way to make it use the skin chooser so I can go through them all to see what I like? My fresh build did not do this and there are too many to put in manually as you have suggested?

chundermike, 2023-01-22:
Next release will scan for all cfg files upon ini file creation and therefore generate a list of all available skins by default

closed 2023-01-22 by chundermike