chundermike/rpi-fruitbox-v2 #30: Re-arrange songs

Presto1970, 2023-01-20:
Hi, is it possible in any skin to manually change the order of the songs shown rather than automatically sorted in alphabetical order and also how many rows/columns etc to relate to buttons I want to fit?

chundermike, 2023-01-20:
Hi,

If you look in the fruitbox.cfg file for the skin you are using there will be one or more entries called "SortSongsBy".  You can set this to several things (see userguide.txt) to sort the songs in different ways.  If you have more than one SortSongsBy entry, fruitbox will do multiple sorts, in the order that they are specified.  For example...

SortSongsBy = year
SortSongsBy = artist

...will sort the songs in year order, then take the result of that sort and sort again by artist.

As for the buttons rows, columns, fruitbox will assign each song on the display with a unique "select code".  The sequence of select codes is determined in the fruitbox.cfg file again, this time using the "SelectButtons" entries.  The number of SelectButtons entries defines the number of digits which make up the select code, and the value of the SelectButtons determines which keys are recognised.  So for example...

SelectButtons = ABCD
SelectButtons = 1234

...will create a list of select codes A1, A2, A3, A4, B1, B2, B3, B4, C1, C2, C3, C4, D1, D2, D3, D4.

If you want the select codes to be 00 to 99, it's...

SelectButtons = 0123456789
SelectButtons = 0123456789

And if you want for example 100 to 399, it's...

SelectButtons = 123
SelectButtons = 0123456789
SelectButtons = 0123456789

Also you can miss out letters; for example some jukeboxes don't have the letter 'I' (as it can be confused with the number one). So in this case you'd have...

SelectButtons = ABCDEFGHJK etc.

If you use the command line option '--debug-song-nums', then fruitbox will display the select code next to each song.  This is useful if you are trying to debug your select buttons...

./fruitbox --skin <skin> --debug-song-nums

Don't forget all this information is documented in the user guide.

Hope this helps,

Cheers,
Mike

Presto1970, 2023-01-20:
Thank you, that’s a great help.

closed 2023-01-21 by chundermike