chundermike/rpi-fruitbox-v2 #35: No sound

Gladzy86, 2023-02-18:
Hello,

Can't wait to get Fruitbox up and running! I'm almost there, I just can't seem to get any of the mp3s to play. They're all listed and I have my buttons mapped but when I select a song, it doesn't play.

Any ideas where I have gone wrong?

Thanks for all your hard work!

Mark

chundermike, 2023-02-19:
Hi Mark,

When you press the selection buttons, does the select code appear in the select code area?

Also, are you running in coin (not freeplay) mode by any chance and haven’t put any credits in?

You could send me the fruitbox.log file and i’ll see if there’s any clues in there...

Cheers,
Mike

bennyboyle, 2024-04-21:
Hi @chundermike, I am also having issues with the sound not playing..
When I open the mp3-files in regular desktop mode, they are playing through VLC.
I have tried in both free play and regular mode. The select code comes up. Other sounds like coin insert and page change are playing. There's just NO response from the actual songs when I punch in the select code. Also trying to press ENTER after I punch in the select code.
Any idea what's up? Running on Raspberry Pi 3B+ with Raspbian GNU/Linux 10 (buster). Installed the recommended version for Pi Zero through 3