chundermike/rpi-fruitbox-v2 #36: can't get IQaudio DigiAMP+ working with fruitbox

Mark-Kiwi, 2023-02-18:
I'm running ver 2.04 of fruitbox on a Pi3 with OS ver 10 (buster) and trying to use a IQaudio DigiAMP+ board for the sound.
Fruitbox works ok when the audio is playing through the HDMI port, but the configuration paperwork for the DigiAMP board says to modify the /boot/config.txt file as follows.
 add
dtoverlay=iqaudio dacplus,unmute_amp
 and comment out the 'dtparam=audio=on' entry to disable the Pi’s on board sound card.
e.g #dtparam=audio=on

I also found references that say to enable the I2C (which makes sense) by removing the # in front of 
dtparam=i2s=on

and to add ' ,noaudio' to prevent the HDMI from transmitting sound

As soon as I do this, save and reboot and run fruitbox, I get the 'error couldn't resolve audio files.

If I re enable the 'dtparam=audio=on' line, reboot etc, It works ok through the hdmi feed (even though I just supposedly disabled that)
Has anyone managed to get the IQaudio DigiAMP+ board working on their Pi3? And if so, how?

Note: I put a clean install of volumio on an SD card and when I boot the pi from that it recognises and works through the DigiAMP board no problems, so I know the board is good.

Help?

chundermike, 2023-02-19:
Have you mapped any buttons in fruitbox to use GPIO?  The DigiAMP board may use GPIO and there could be some conflict there perhaps?

Mark-Kiwi, 2023-02-20:
(My second attempt at typing this.....fxx**ing computer crashes!)
Hey mate!
So just in case I had done something to create this issue for myself, I decided to start again with a clean install on a different SD card.
Followed the readme instructions, installed a new buster lite version, used sudo raspi-config to increase GPU memory to 256, enabled SSH and WiFi and also enabled I2C.
Installed fruitbox as per steps 3,4,5.
Used WinSCP to create a Music folder and dropped some music in it, also added the classic skin to the list in the fruitbox.ini file so I could have a choice of skins.
Even gave it a reboot.

Ran fruitbox, it works thru the hdmi sound perfectly!

Then modified the /boot/config.txt file as per the IQAudio digiAMP instructions....
ie added the line
dtoverlay=iqaudio dacplus,unmute_amp
and commented out the dtparam=audio=on to disable the Pi's on-board sound card.

I also checked the I2C had been enabled

found this...
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on
so yup, looks good. 

Note: the video guide I watched previously on youtube (How To Set Up An IQaudIO DAC+ With Volumio and Raspberry Pi OS) said to uncomment dtparam=i2s=on, not the dtparam=i2c_arm=on... I think that was a mistake but I have tried both...)

I rebooted and ran fruitbox. When I select the skin, I get the 'ERROR couldn't reserve audio samples'
(sorry I mis-spelt the error in my earlier description)

Now this is where it gets interesting...
with 'dtparam=audio=on' enabled, when I use raspi-config,  I see audio options as
0 HDMI 1
1 Headphones

but I don't see the audio DAC card

But when I run alsamixer, I see in the top left corner that the 'card' is bcm2835 HDMI1 

The HDMI1 control is  displayed on screen, and when I go to sources, I see
-default
0 bcm2835 HDMI 1
1 bcm2835 Headphones
2 IQaudIODAC

so it sees my card, it's just not the default card. The default is the HDMI card.

When I run 'speaker-test -c2 -t wav' (something I discovered in the internet) I get the audio test through the HDMI port.

Now if I go back and comment out 'dtparam=audio=on' in the /boot/config.txt file and reboot...then look at raspi-config..there are 'no audio internal audio devices found'.

and when I run alsamixer, the default is the IQaudIODAC card. There are no other sources.
NOW when I run 'speaker-test -c2 -t wav' the test audio booms out the speakers...thus testing the board works AND it's not muted. (I knew it wasn't muted because the mute light is off)

Now if I could just get it working on fruitbox!

Re GPIO pins, the paperwork has the following info...maybe it helps...maybe not...I still think it's something to do with making the DAC card the default audio...maybe if there was some way to make it appear when I run the raspi-config command...

_We expect the following Raspberry Pi GPIO to be used by IQaudio audio boards
ALL POWER PINS
ALL GROUND PINS
GPIO 2/3 (I2C)
GPIO 18/19/20/21 (I2S)

If appropriate then the following are also used
GPIO 22 ( mute unmute support)
GPIO 23/24 for Rotary Encoder (physical volume control) or status LED (Codec Zero)
GPIO 25 for the IR Sensor.
GPIO 27 for the Rotary Encoder push switch Code Zero switch
_

looking at this....maybe I need to enable the I2S as well?
tried enabling....and no, still getting the error couldn't reserve audio samples.

Mark-Kiwi, 2023-02-22:
Hey Mate,
I've tried a clean install and still getting the same issue. (I've posted
another comment explaining what I've done and discovered.)
Basically It looks like the sound board is set up correctly, it's just when
I disable the onboard sound board to make the sound appear on the digiAMP
board, the fruitbox software won't recognise that available soundboard.

Cheers!
Mark

Mark-Kiwi, 2023-02-22:
I do a little bit of coding of arduino's but this is way beyond me...but I tried hunting for the error message in the code.
(What language/environment is this coded in?)

In the config.cpp file, on line 438 I found the error message. "Couldn't reserve audio samples"

The problem seems to be  related to these lines...
  
 _// Add extra sound channels for camera shutter sound effect...
  StatusDisplayClass::num_sound_channels++;

  if (!al_reserve_samples(StatusDisplayClass::num_sound_channels)) error("Couldn't reserve audio samples");_

chundermike, 2023-02-22:
Hi Mark

Fruitbox is written in C++. The error comes from the Allegro library I'm using, so i can't be sure why it's complaining. Usually with this sort of stuff it's OS related, rather than fruitbox itself.

Not much help for you I'm afraid :(

Mike

Mark-Kiwi, 2023-02-23:
Hey Mike,
I just tried using the version 1 of fruitbox by burning your image to SD card then modifying the /boot/config.txt to disable the on-board sound card and enabling the digiAMP+ card ....and it works!

Obviously not the ideal soultion as I want to be using the version 2 of fruitbox.
I wonder, are not other people using sound cards? or is it just this model sound card with this problem? Surely people are wanting the best sound from such an awesome program!
Maybe there is a different sound card I could buy? Any recommendations on one you know works?

While trying to compare build versions, I see that it looks like fruitbox ver 1 uses allegro-5.2.4.0, where-as ver 2 uses allegro-5.2.5.0 ..... Could the issue be as simple as that? Maybe just needs compiling using the older version of allegro?....or maybe there is a newer version of allegro? (I don't know what allegro does, all I can figure out is that it's some kind of library...I think :-) ) 

Not sure I have the skills or knowledge to nut my way thru recompiling this with different libraries or understanding the consequenses or dechyphering error messages etc..

Thoughts?

Cheers!

Mark

chundermike, 2023-02-24:
Hi Mark

That's interesting...thanks for your input. I'll do a fruitbox v2 build with the older Allegro version to see if it fixes your problem.

What pi are you using? If pi4, are you on 32 or 64 bit Desktop?

Cheers
Mike

chundermike, 2023-02-24:
Hi again

It more likely that it's the Raspbian version not Allegro version. The V1 SD image is quite old.

What OS version were you using with v2?

Mike

Mark-Kiwi, 2023-02-25:
Hey Mike,
I'm using a Pi 3 Model B+

(hopefully will upgrade to a Pi 4 once they become available again...and at
a reasonable price)

And started with the 2022-01-28-raspios-buster-armhf-lite.zip using the
link from your https://github.com/chundermike/rpi-fruitbox-v2 page.

Cheers!

Mark-Kiwi, 2023-02-28:
Hey Mike,
as an experiment, I used the ver 1 image (so I had the old version of OS that worked under 1), deleted the fruitbox v1 folder, then installed the Ver 2.
As soon as I alter the /boot/config.txt file to disable the onboard sound, the problem reoccurs.

:-(

chundermike, 2023-03-02:
> Hey Mike,
> I started again with 2022-01-28-raspios-buster-armhf-lite.zip and followed along with the build instructions and was able to compile ok. I then made the relevant changes to the config.txt file and sure enough I had the same issue.
> With the dtparam=audio=on enabled, I get sound on the monitor through the HDMI connection, so I know my compile was good.
>
> I then started again but this time before compiling I opened up the config.cpp file and commented out the line
> if (!al_reserve_samples(Config->num_sound_channels)) error("Couldn't reserve audio samples");
> Just in case I could get away with just removing the line that displays the error.
>
> While the compile seemed to go okay, the program crashed when I tried to run it, so obviously the line is more critical than just
> displaying an error message if ceertain conditions exist.
>
> Did you manage to compile it with an older Allegro version?
>
> Or have any other thoughts on what this part of the code does or how to fix it?

Hi Mark

Yes, that line you removed is critical!  It's a call to one of the Allegro audio functions to tell it how many sound channels you need. The fact that it returns an error is something to do with the OS, so out of fruitbox's control :(

I haven't managed to do a build yet with a previous Allegro version. If I get the time soon I'll do it. If you are in a hurry though you could always try it yourself...you've done the build before so know what you're doing. All you need to do is download an earlier version of Allegro as a zip file (a bit of googling will help you to find it), then just follow the build instructions as before, but using the older Allegro zip file.

Cheers
Mike

Mark-Kiwi, 2023-03-13:
Hey Mike,
Well I managed to compile ver 2.04 using Allegro 5.2.4.0 and while it compiles ok and runs when the onboard audio is enabled in /boot/config.txt, as soon as I disable it by commenting it out, I get the same issue. (The 'ERROR couldn't reserve audio samples' problem and it doesn't work through the Audio board.

I double checked and the version 1.16 works ok.

I've tried examining the code but I can't make head nor tail of how it's coded.

I did try modifying the config.cpp file to see if I could make it work a bit like the ver 1.16.
ie commented out line 436  
StatusDisplayClass::num_sound_channels++;

and added line 305 from ver 1.16
Config->num_sound_channels += StatusDisplayClass::num_sound_channels;

I found I had to add the following to the config.hpp
uint32_t num_sound_channels { 7 }; /* mp3, coin, page move, load_song, unload_song, invalid song, song video ... video sound channels are added when video onjects are defined */

(which was copied from the ver 1.16 config.hpp)

While it compiled, I couldn't make the program run. (I can't remember exactly what it did or didn't do...I've tried so many itterations I loosing track of things.)

To me it appears that versions 1.16 and 2.04 handle things completely differently but I don't understand the code or how it functions to diagnose it.
I've already tried compiling 2.04 on the imageof version 1.16 and that didnt solve the issues so I'm lost and way out of my depth.
Any thoughts?

chundermike, 2023-03-13:
Hi Mark

Sorry you're spending a lot of time on this. When I get the chance, I'll have a look myself and see what the differences are in the audio between v1 and v2 and see if I can spot something.

Cheers
Mike

Mark-Kiwi, 2023-04-19:
Hey Mike,
Have you had a chance to figure out why V1 works but V2 doesn't?

Cheers!
Mark

chundermike, 2023-04-19:
Hi Mark,

Really sorry, I’ve not managed to take a look yet. It’s still on my list though.

Cheers
Mike

Kipper641, 2023-05-06:
Apologies for jumping in,

Having what appears to be exactly the same problems using an INNOMAKER amp hat.

Works perfectly on v1 

Get "couldn`t reserve audio samples"

Many Thanks

andre, 2023-05-24:
Hello, I also wanted to add that I am getting the `Error: Couldn't reserve audio samples` with:

* Fruitbox v2
* Raspberry pi 3b
* HiFiBerry Amp2

`aplay -l` output is good and audio playback works through HiFiBerry with mplayer. But with Fruitbox I run up against that `Error: Couldn't reserve audio samples`

chundermike, 2023-05-24:
Hi

There was an issue with the Allegro library for some versions of Debian regarding audio. I can’t remember off the top of my head but somewhere in the readme.md instructions it mentions which OS to avoid. Maybe try a different version of Raspbian? Long term,I will try and come up with a better solution.

Mike