chundermike/rpi-fruitbox-v2 #41: Autorun Help

mpaptyltd, 2023-06-11:
Hello could someone please write a clear and concise guide on correctly how to run this on AutoStart on power up
I have looked and there is so many ways to do this and all are very confusing for people new to this
Even in the guide where it says how to do it, it so vague a two line sentence, it appears im not the only person pulling there hair out with this and none of the listed answers work 
Surely somebody has a simple solution for this

chundermike, 2023-06-11:
Hi,

Sorry you're finding this confusing. I tried to make the instructions clear but appreciate that it's difficult sometimes if you have knowledge of something to explain it to someone without that knowledge to understand easily.

There's a chap called Jay Thompson who has written a dummies guide for fruitbox installation which you may find helpful. He has a Discord channel called "Jay's Arcade", with a sub channel purely for the dummies guide. Alternatively he has posted the guides on the Facebook "Digital Jukebox Builders" group.

Cheers
Mike

mpaptyltd, 2023-06-12:
PROBLEM FIXED BY BELOW
[Uploading Pi4 Fruitbox V2 GUI Install Guide for dummies.pdf…]()

closed 2023-06-12 by mpaptyltd