chundermike/rpi-fruitbox-v2 #47: Unable to install fruitbox v2 on raspberry pi4 64 bit

breikers, 2023-08-03:
Hi Mike, when i try to install fruitbox on my new Raspberry Pi4 I get the following message after ./install_fruitbox_pi4:  No such file or directory. Any ideas what is going wrong here?

Regards

Jack

cjthedj70, 2023-12-05:
I am having the exact same problem...HELP!!!      TIA

cjthedj70, 2023-12-06:
I solved this problem for me.  I had my Pi 3 ready to load Fruitbox, then read that I needed a Pi 4 instead.  After I got the Pi 4, I just moved the SD Micro card from the Pi 3 to the Pi 4.

I then started over and reloaded the Pi OS onto the micro card, specifying it was for a Pi 4.  That solved the problem of not being able to execute the program.  YMMV