chundermike/rpi-fruitbox-v2 #51: Skins not filing the screen and "cut off" at the top. (RPI4 8gig PI OS 64bit)

Mike-002, 2023-11-20:
Sort of new to the RPI4 and latest PI OS 64 bit.
I have tried two different PIs (Identical) each with its own 32gig sd. The HDMI monitor is a dell 24". I have the desktop configured for 1920x1080. When I launch Fruitbox the result is the skin doesn't fill the screen completely. There is a gap at the top and at the bottom where I can see the tool bar and the desktop background. The top of the skin appears to be clipped too.

I have been looking on the interwebs for a solution but I don't see anyone having this issue. Make me feel special but confused at the same time :)

I will see if I can grab a screenshot so there is no confusion as to what I am experiencing. 
[image]

DereIBims, 2023-12-10:
I have exactly the same problem

Mike-002, 2023-12-13:
@DereIBims I ended up installing Bullseye 64bit os and then the option to hide the taskbar was there to use.

I guess we just need help with bookworm to find that setting wherever it got buried :)

DereIBims, 2023-12-13:
Oh nice, thanks!
I'll try that for sure

jaythom1, 2024-05-01:
Bookworm with the Wayland backend does not give you an easy way to hide the menu bar. It is possible however. There are two ways to do it. 

Option one: Switch the backend to use X11 (the option to switch can be found in raspi-config). Once you do this, you should be able to right click the menu bar and click Panel Settings. Click Advanced and check the Minimize panel when not in use check box. Under size when minimized, set it to zero pixels. Using X11 however defeats the purpose of using Bookworm.

Option two: To autohide the menu bar in the Wayland back end on Bookworm you'll need to edit the wf-panel-pi.ini file. In a terminal window, type `nano /home/pi/.config/wf-panel-pi.ini` and press enter. Inside the editor, add to the bottom of the file…

```
autohide=true
minimal_height=45
autohide_duration=300
```

Note: this can break widget functionality if you have any widgets running on the taskbar.

This tip and more can be found in the Fruitbox V2 Dummies guide. https://fruitbox4dummies.com/pi4pi5/#15-hide-the-menu-bar-recommended-raspberry-pi-os-bookworm-with-the-default-wayland-backend