chundermike/rpi-fruitbox-v2 #52: Database in wrong loation

NJW66, 2023-12-01:
Hello Mike, fantastic piece of software.
However I am having a problem, fruitbox seems to write the database for my songs on my USB in the wrong location and subsequently cant find the songs.
I have a USB stick with 300 ish songs in the MUSIC directory and  10 Albums in the Albums directory. Fruitbox only writes 56 songs to the MUSIC fruitbox.db, then writes all of the songs into the correct Albums "fruitbox.db" but then it writes the remaining 244 songs from the MUSIC directory into a random Albums .db. And as such cant find them when I launch Fruitbox although they show up on the screen
I cant figure out why its happening, any help would be much appreciated

Dex-too, 2023-12-19:
I just had this problem too. Usually I have no music directly in the configured music folder (everything is in a subfolder in some way), but this time I set up a Christmas jukebox with a bunch of songs in the root folder, but also with a handful of subfolders of music. What I found was that Fruitbox found all of the music, and wrote proper db files in each of the subfolders, but would not write a db file in the root folder. It instead stuffed all of the music it had found in the root folder into each of the subfolder dbs, in random batches (so one subfolder had about half of the root music, another had a quarter, another had one song, the next subfolder had no songs, and the final subfolder all of the remaining songs). This caused them to show up in the Fruitbox UI, but whenever anybody selected them, they wouldn't play. When I later realized what was going on with the scattering of entries into the subfolder dbs, I manually edited each subfolder db to cut out the extra songs listed there and paste them into a db in the root folder that I created. After that Fruitbox worked exactly as expected.

I deleted all of the dbs and let it rescan and recreate everything and it did it again. I created a different root folder and dropped some music in it and in some subfolders too, and after it scanned everything on startup it did the same thing there.

chundermike, 2023-12-19:
Thanks for reporting this, and the detail you've gone into. I will add it to my bug list and look into it.
Cheers
Mike