chundermike/rpi-fruitbox-v2 #53: Got It Working!

cjthedj70, 2023-12-06:
Noticed that while playing, the video clip that shows the "record" playing has the tone arm on the wrong side.  Any way to move it to the right side in the image?

chundermike, 2023-12-06:
Yes, in the skin cfg file it will be generated by a status object, so just modify its position (or flip it)

chundermike, 2023-12-06:
To get touchscreen working you need to enable it in the ini file and also use/create a skin which has touch buttons defined. Some of the skins in GitHub have this already :)

Thanks for the quick reply!

My next steps are to set it up to work with the touchscreen. I think that involves editing fruitbox.ini, looking forward to reading about it.

Sent from [Proton Mail](https://proton.me/mail/home) for iOS