chundermike/rpi-fruitbox-v2 #54: Screen size and search question

cjthedj70, 2023-12-09:
I am using a SunFounder 10" touchscreen.  When fruitbox loads, the fruitbox graphic is about 1/8 inch below the underlying Raspberry Pi desktop graphic.  I think the bottom of the fruitbox graphic is also slightly higher.  How can I "stretch" that graphic slightly to fill the screen?

Also, I only find three skins that work with Touchscreen; AlbumPlayer, TouchMiniAlbums, and TouchBlue.  Are there any plans to convert any of the other skins to touchscreen use?  How can I learn how to do that myself or create my own touchscreen enabled skins?

On the three skins that do work with touchscreen, I cannot find a search function.  With over 6000 songs on my USB stick, I will need to be able to search by artist, song, album, etc for this jukebox to be functional.

Thanks for your support!

chundermike, 2023-12-09:
Hi

Not sure about the stretch…maybe worth trying the —display-size command line parameter and also look at the dummies guide (google to find that). You can create or modify skins to use touch screen…you’ll need to read the user guide carefully but basically you need to define touch areas using the [touch buttons] group. For the search function, hit forward slash on the keyboard to open the search menu.

Mike

cjthedj70, 2023-12-11:
I think I misstated the issue.  The top of the fruitbox screen is actually cut off a bit.  For example, using the AlbumPlayer skin, the albums on the top row are cut off slightly.

This screen has 1200x 800 resolution.  I tried using—display-size 1200 800, but got ERROR Failed to create 1209 x 899 display.

cjthedj70, 2023-12-12:
Still trying to solve this problem.  I'm a little OCD, so I just can't settle for half of the Raspberry Pi task bar showing at the top of the jukebox.

cjthedj70, 2023-12-13:
Got the screen size issue corrected, thanks to "Pi4 Fruitbox V2 GUI Install Guide for dummies 1.6" by Jay Thompson.  Only bad part is I had to create a Facebook account to download the file.

My goal now is to find a skin that works with touchscreen, and includes a search button in the skin so I don't have to use the keyboard at all.

Can you point me in the right direction as to, for any particular skin, which file I need to modify to reassign a current function to the search function?

Thanks for all the support, like how this project is progressing.

chundermike, 2023-12-13:
That’s good news. For touch skins, just search the fruitbox.cfg files and look for [touch buttons]. These define touch areas for buttons. You can add another for your search button. Read the user guide for details about this.

Mike-002, 2023-12-13:
> Got the screen size issue corrected, thanks to "Pi4 Fruitbox V2 GUI Install Guide for dummies 1.6" by Jay Thompson. Only bad part is I had to create a Facebook account to download the file.
>
> My goal [...]

I haven't been able to find this "Guide for dummies"
Can a link be shared?
I ended up reverting to Bullseye 64 bit to be able to hide the taskbar solving my fullscreen issue.

closed 2024-01-17 by chundermike