chundermike/rpi-fruitbox-v2 #59: Support for 4x3 and 4x4 keypads

Harry-Badger, 2024-06-18:
Great program and I have built a wall mounted Juke Box. 
I'm trying to get fruitbox to work with a 4x4 or 4x3 keypad.
If I've understood fruitbox correctly there is a one-to-one GPIO + GND mapping for each key?
16 key 4x4 keypads have 8 pins that are mapped to rows and columns and there is not a common GND.
Is it possible to use such a keypad with fruitbox?

chundermike, 2024-06-19:
Hi
That type of keypad needs a scan method to read the keys which fruitbox doesn't support I'm afraid. Maybe there is a third party app which can do this which will run alongside fruitbox and provide it with the key codes?
Mike

Harry-Badger, 2024-06-19:
Thanks Mike. Understood.
I've decided to scrap the 4x4 keypad idea as the script to constantly scan and cater for debounce was getting too complicated. 
Instead I am opting for mechanical MX keyboard switches with bespoke resin printed keycaps.  Took the inspiration from this : https://www.printables.com/model/140766-16-keys-macropad

closed 2024-06-19 by Harry-Badger