chundermike/rpi-fruitbox-v2 #64: Limited tracks

annibale666, 2024-10-11:
Hello
I use the SB-M100 theme and have the possibility to have only two screens with 32 tracks each. The left-screen and right-screen keys skip only once. I currently have 120 songs in the Music folder but it only displays 32 songs on each screen, so 64 songs.
If I use the BigBobsMillions theme instead, I have all 120 of my songs on the screen.

Can we increase the amount of screens to browse left and right, or is it really just these two?
Do the themes have a track display limit?

Thank you and have a great day.

chundermike, 2024-10-11:
Hi

Fruitbox will create as many screens as it needs to accommodate all your songs. The number of songs per screen is determined by the skin itself and is the number of title strips specified in the skin config file.

Hope this answers your question?

Cheers
Mike

chundermike, 2024-10-11:
Maybe I misunderstood your question… there are three ways to move through screens…
1. Left/right will move one screen at a time
2. Left screen / right screen will move X screens, where X is defined in the ini file (or cfg file I can’t remember, and also I can’t remember the name of the parameter, sorry)
3. Left sort / right sort will jump to the next alphabetical letter in the last sort criteria

Cheers
Mike